vlachoudis/bCNC #1422: No circular motion

keith124-code, 2020-06-14:
Running bcnc on rpi4b have homing and linear motion, but system is not responding to gcode for circular motion ie: G2, G3, G12 and G13, have you any thoughts on cause or have I missed something.

MARIOBASZ, 2020-06-14:
I don't use rpi.
I understand that grbl supports G2 G3, not G12 G13
Is the code writing well?
R or I J?
If R try with angle less than or equal to 180 degrees (I don't remember correctly, if the limit is 180 degrees)
grbl must be sending the corresponding error number.

keith124-code, 2020-06-14:
Have tried gcode programs using g3 and g2, linear motion works but program stops at  g2 or g3 code, have run program on simulator and it works fine!

MARIOBASZ, 2020-06-14:
can you copy the code?

keith124-code, 2020-06-15:
Morning, this is the basic program, runs on the Gcodesim, only does linear motion then stops. Machine will run long linear progs just not the circular motion, is it possible there is an error in the down load or am I just missing something. I am not a programmer have only used Denford Triac mill system Gcode from 1981 just trying to get into CNC, this is the code I have been trying to run, any help would be well as to the issue.Many thanksKeith ;hole cutting prog for millg01 x0 y0 z1 f250m3g01 x-9.5 y0 z-0.25 F50g02 x-9.5 y-9.5 i0 j0.95 F50g01 x-9.5 y0 z-0.5 F50g02 x-9.5 y-9.5 i0 j0.95 F50g01 x-9.5 y0 z-0.75 F50g02 x-9.5 y-9.5 i0 j0.95 F50g01 x-9.5 y0 z-1.0 F50g02 x-9.5 y-9.5 i0 j0.95 F50g01 x-9.5 y0 z-1.25 F50g02 x-9.5 y-9.5 i0 j0.95 F50g01 x-9.5 y0 z-1.5 F50g02 x-9.5 y-9.5 i0 j0.95 F50g01 x-9.5 y0 z-1.75 F50g02 x-9.5 y-9.5 i0 j0.95 F50g01 x-9.5 y0 z-2.0 F50g02 x-9.5 y-9.5 i0 j0.95 F50g01 x-9.5 y0 z-2.25 F50g02 x-9.5 y-9.5 i0 j0.95 F50g01 x-9.5 y0 z-2.50 F50g02 x-9.5 y-9.5 i0 j0.95 F50g01 x-9.5 y0 z-2.75 F50g02 x-9.5 y-9.5 i0 j0.95 F50g01 x-9.5 y0 z-3.0 F50g02 x-9.5 y-9.5 i0 j0.95 F50g01 x-9.5 y0 z-3.1 F50g02 x-9.5 y-9.5 i0 j0.95 F50g00 z20 f500m5m2 Sent from Mail for Windows 10 From: Mario BaszSent: 14 June 2020 22:42To: vlachoudis/bCNCCc: keith124-code; AuthorSubject: Re: [vlachoudis/bCNC] No circular motion (#1422) can you copy the code?

PatchTech, 2020-06-15:
Having a similar problem. On a G02 command, cutter moves from <last position> straight to the coordinates specified as the center of the arc and then to the coordinates specified by the next command. Editor view shows the same functionality. Code as follows:-
%
G21
G90
G94
F140.00
G00 Z2.0000
G17 M3 S3000
G4 P1
G00 X27.3250Y6.1270
G01 Z-1.0000
G02 X21.3250Y0.1270X21.3250Y6.1270F100.00
G01 X5.0000Y0.1270
G00 Z2.0000
G00 X0Y0
M05
%
I'm running version 0.9.14-dev of cBNC under Python2.7 on a RPi3B with the Raspbian Buster OS.
If it'll help I can post a screenshot of the editor view.

PatchTech, 2020-06-15:
r.e. previous post. I've just spotted an error in my code, Line 11 should read 
G03 X21.3250Y0.1270I21.3250J6.1270F100.00
However router functionality remains the same, Looking closely at the status bar I noticed that Error:33 flashed very briefly - does that mean that the router doesn't support G02/G03 commands?

PatchTech, 2020-06-16:
Please ignore previous posts as I now have the answer; I & J are RELATIVE coordinates not absolute (that was not made clear in the tutorial I was reading) Line 11 should read:-
G02 X21.3250Y0.1270 I-6.0000J0.0000 F100.00
Router & bCNC now working correctly.
Other Noobs BEWARE!

keith124-code, 2020-06-16:
PatchTech, Mario Basz Have tried Patch Tech prog, runs on gcodesim fine, tried on Triac mill with RPi4b just draws the linear motion,Thanks for your help, any other ideaskeith Sent from Mail for Windows 10 From: PatchTechSent: 16 June 2020 01:04To: vlachoudis/bCNCCc: keith124-code; AuthorSubject: Re: [vlachoudis/bCNC] No circular motion (#1422) Please ignore previous posts as I now have the answer; I & J are RELATIVE coordinates not absolute (that was not made clear in the tutorial I was reading) Line 11 should read:-G02 X21.3250Y0.1270 I-6.0000J0.0000 F100.00Router & bCNC now working correctly.Other Noobs BEWARE!