vlachoudis/bCNC #1423: Skew compensation

deirdreobyrne, 2020-06-15:
I have a cheap CNC machine where it is pretty impossible to get X and Y orthogonal to each other. I would like to use it to make two-sided PCBs, which requires the PCB to be flipped and the co-ordinate system re-established on the other side. That cannot happen when X and Y are not orthogonal.

It's been mentioned before that any software correction should be done by grbl. However the grbl developers point out that the 8-bit microcontrollers that their software tends to run on is not powerful enough to do the calculations on-the-fly, and suggest that the calculations should be done before the gcode is sent to grbl.

Is there ANY chance bCNC could implement a fix for a skewed X and Y axis? As the owner of a cheap CNC machine I'd be prepared to measure my skew without any extra help from bCNC, but I don't know how to alter the code so that it sends corrected co-ordinates to grbl.

mbaaba, 2020-06-15:
If you know the diff, you could patch your gcode-file

MARIOBASZ, 2020-06-15:
I think I have seen that in grbl there is a branch where they have added correction for squaring and also the same or another one for backlash. It is the ideal place.
@mbaaba, I think it requires a little more work.
I understand that there are at least two ways: one is with a plugin that modifies the gcode (from my experience it is not easy).
The ideal option is another, and that is that it forwards the modified gcode always according to orthogonality and backlash correction values.
My question is if it only works for straight lines.
Anyway, better if you can adjust the portico (one of the advantages of double motor in Y axis)

deirdreobyrne, 2020-06-15:
> If you know the diff, you could patch your gcode-file

I presume you are talking about sending the gcode through a parser, and then through a simple algorithm to do the skew. Do you know of any general-purpose parsers out there? My programming skills in modern languages isn't great - it would have to be easily-modified. Thanks.

deirdreobyrne, 2020-06-15:
> I think I have seen that in grbl there is a branch where they have added correction for squaring and also the same or another one for backlash. It is the ideal place.

Do you know which branch that is?

> I understand that there are at least two ways: one is with a plugin that modifies the gcode (from my experience it is not easy).
> The ideal option is another, and that is that it forwards the [...]

bCNC already modifies the gcode when the "Orient" feature is used, and it seems to work for circles. I don't know python - might it be a simple change to already-existing code which implements "Orient"?

MARIOBASZ, 2020-06-15:
Orient I have not used it, is it not to orient the camera?
Maybe you use scaling, distort Y with respect to X?
if it works can you let me know?

deirdreobyrne, 2020-06-16:
> Orient I have not used it, is it not to orient the camera?

I use it to correct for the fact that when I flip a PCB to the other side, I need to correct the co-ordinate system to have a new origin and a new angle between the X and Y of the machine and the X and Y of this side of my PCB. I drill four small holes in a square on one side of my PCB, flip it, measure where the holes are, and use those measurements to enable bCNC to shift and rotate the gcode so as to carve traces in the right position.

MARIOBASZ, 2020-06-21:
Have you found a solution?
Meanwhile, use Rotate (if you don't specify center, rotate from 0; 0)?

deirdreobyrne, 2020-06-21:
> Have you found a solution?

I believe I have - outlined above. However my programming skills are not up to the task of implementing it.

> Meanwhile, use Rotate (if you don't specify center, rotate from 0; 0)?
> I use it from the command line.

How will rotate cure a skew between the X and Y axes? Is it possible to rotate just one axis independent of the other?!

MARIOBASZ, 2020-06-21:
.... decidedly sometimes my intelligence amazes me ...: rotating a rhombus results in a rotated rhombus, not a square !!!
Orienting works?

deirdreobyrne, 2020-06-21:
The solution - it seems to me that the code which implements "Orient" should be easily modifiable to implement skew correction. The only difference I can see between what "Orient" does (successfully) and what skew needs to do is the maths. Unfortunately seeing if that is the case, and implementing it if it is, is beyond my skills.

MARIOBASZ, 2020-06-21:
perhaps this way can result in the best solution.

https://github.com/gnea/grbl/pull/709

MARIOBASZ, 2020-06-22:
if it works satisfactorily, i will let you know

deirdreobyrne, 2020-06-22:
Thank you - I didn't see that change request for grbl. I'm awaiting delivery of an accurate square with which to do my measurement - I'll let you know.

deirdreobyrne, 2020-06-25:
The accurate square I got has now (finally!) squared my machine to more than a sufficient degree of accuracy. The grbl patch you've linked to above seems quite simple and, according to what I can see in the comments in the rest of grbl, it should work. I've installed it on my machine, and I might just run a few tests with an artificially high skew and see how it reacts. But for now I think this matter can be considered closed.

closed 2020-06-25 by deirdreobyrne