vlachoudis/bCNC #1426: Not connecting to Arduino Mega on latest Version

dylanpil, 2020-06-18:
Previously I had bCNC-0.9.14.304 running and connecting to COM port (Tibbo virtual serial port) and arduino Mega.
I am using an Arduino mega board with wifi built in (ESP8266 on board).
I am using grbl fork for MEGA from:
https://github.com/LETARTARE/grblQ-Mega-4axes
I have installed the latest version of bCNC and tried both installing with python 2.7 & 3.
Both installs connect to a std COM, but not to the arduino Mega via usb or wifi. I have 2 arduino megas that I have tested with no luck.
I would love to get this working with my 4 axis cnc. Please advise

Below is what i can find in the python terminal:

COM20   Tibbo Virtual Serial Port (COM20)       ROOT\*TVSP\0000
COM1    Communications Port (COM1)      ACPI\PNP0501\1
COM17   Arduino Mega 2560 (COM17)       USB VID:PID=2341:0042 SER=557353034343511091F1 LOCATION=1-1.3
COM3    Arduino Uno (COM3)      USB VID:PID=2341:0043 SER=8543034303835160A020 LOCATION=1-1.2
comport fix
Controller state changed to: Alarm (Running: False)
comport fix
Controller state changed to: Idle (Running: False)
Exception in thread Thread-4:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "C:\Users\dylan\Downloads\bCNC-master\bCNC-master\bCNC\Sender.py", line 777, in serialIO
    elif self.mcontrol.parseLine(line, cline, sline):
  File "C:\Users\dylan\Downloads\bCNC-master\bCNC-master\bCNC\controllers\_GenericController.py", line 201, in parseLine
    self.parseBracketSquare(line)
  File "C:\Users\dylan\Downloads\bCNC-master\bCNC-master\bCNC\controllers\GRBL1.py", line 217, in parseBracketSquare
    CNC.vars["G92B"] = float(word[5])
IndexError: list index out of range

Below is from the serial_log.txt:
[serial_log.txt](https://github.com/vlachoudis/bCNC/files/4796405/serial_log.txt)

Grbl 1.1e ['$' for help]

[MSG:'$H'|'$X' to unlock]

<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000,0.000>

?<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0|Ov:100,100,100>

?<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0>

?<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0>

$X
[MSG:Caution: Unlocked]

ok

?<Idle|MPos:0.000,0.000,0.000,0.000|FS:0,0>

$#
$G
[G54:0.000,0.000,0.000,0.000]

[G55:0.000,0.000,0.000,0.000]

[G56:0.000,0.000,0.000,0.000]

[G57:0.000,0.000,0.000,0.000]

[G58:0.000,0.000,0.000,0.000]

[G59:0.000,0.000,0.000,0.000]

[G28:0.000,0.000,0.000,0.000]

[G30:0.000,0.000,0.000,0.000]

[G92:0.000,0.000,0.000,0.000]

!!

MARIOBASZ, 2020-06-21:
You load an arduino Uno and does it work?
Maybe the same error that happened to me:
The problem was in Arduino files, having two files with the same name, did not evise an error but Mega was dumb.
Using Web, the solution was to remove grbl from the directory to another site so as not to lose it.
I already have my Mega 6 axes to use.

dylanpil, 2020-06-24:
@MARIOBASZ 
arduino Uno runs fine, 
It could be the version im using.
I am using grbl fork for MEGA from:
https://github.com/LETARTARE/grblQ-Mega-4axes
I haven't had the chance to try flash a different Mega grbl, as I assume that if the Uno works, it would probably work on a different fork.

MARIOBASZ, 2020-06-24:
what is the fault? Arduino Mega responds or does not respond at all: in other words, at least it reports grbl 1.1 when connecting, bCNC is listed as not connected, however it moves the axes (no solution known)? Or conversely there is no answer at all? (this error is what happened to me and I solved it)
Try another sender to see if the problem is with bCNC or firmware

MARIOBASZ, 2020-06-24:
with another sender does it work?
my knowledge is limited.
press stop, press delete alarm, press homming?

Harvie, 2020-06-24:
This seems to me like it is trying to parse 6-axis report, but only 4 axes are reported... `CNC.vars["G92B"] = float(word[5])`
We need to fix the code so it only parses the axes sent by grbl and ignore missing axes.

Harvie, 2020-06-24:
As workaround you might try to configure your arduino as 6axis (but connect only motors for 4axes)

Harvie, 2020-06-24:
take a look at this code in GRBL1.py:

```python
		if word[0] == "G92":
			CNC.vars["G92X"] = float(word[1])
			CNC.vars["G92Y"] = float(word[2])
			CNC.vars["G92Z"] = float(word[3])
			#if Utils.config.get("bCNC","enable6axis") == "true":
			if len(word) > 4:
				CNC.vars["G92A"] = float(word[4])
				CNC.vars["G92B"] = float(word[5])
				CNC.vars["G92C"] = float(word[6])
```

in several places there is something like `if len(word) > 4` which tests if the input is 3axis or 6axis.
We need to rewrite this into more generic code, which autodetects the number of axes, whether it is 1 or 99.

tsmarks, 2020-06-25:
I think the solution should be fairly simple.  There are three places where the code looks similar to:
if len(word) > 4:
				CNC.vars["G92A"] = float(word[4])
				CNC.vars["G92B"] = float(word[5])
				CNC.vars["G92C"] = float(word[6])

if we change all three to something like:
if len(word) > 4:
				CNC.vars["G92A"] = float(word[4])
if len(word) > 5:
				CNC.vars["G92B"] = float(word[5])
if len(word) > [6:](url)
				CNC.vars["G92C"] = float(word[6])

That should enable 4 and 5 axis systems.  I am more than happy to create a test GRBL1.py if someone is wiling to test it.
[GRBL1.py.txt](https://github.com/vlachoudis/bCNC/files/4828627/GRBL1.py.txt)
Just rename it to GRBL1.py and save it over the original.  I don't have a 4 or 5 axis controller right now to test this.

dylanpil, 2020-07-02:
THANKS, I haven't had a chance to test, but i'll see if i can in the next week

Harvie, 2020-07-02:
If it works, please make PR...

Netzradio, 2020-08-24:
I have a MPCNC with dual X& Y Axsis. I tested the patched file from tsmarks and it work for me.
Thanks a lot.

bosd, 2020-09-13:
I see this is merged in the master repo.
but by using the pip installer the old grbl1 file was installed.
So, manually installed the patched file of @tsmarks .

Edit: Inittially it did not work. I had to restart bCNC to get it to work. (stupid mistake) Now it is working fine.

I am using an Arduino Mega with onboard wifi and esp, with grbl-mega-5x.

Harvie, 2020-09-14:
> I see this is merged in the master repo.
> but by using the pip installer the old grbl1 file was installed.

I've uploaded master to pypi, so it should be in sync now.

bosd, 2020-09-15:
However, I do have a problem with the autolevel function. Don't know if it is related. 
The program freezes at the last probe point. (status remains run, but console is unresponsive)
My skills are not on par to debug it.

closed 2020-09-16 by Harvie