vlachoudis/bCNC #1427: CNC 1310 autolevel issue

jorge-chavez, 2020-06-18:
I've been working with bCNC for about a month and works great, but I want to evaluate my chinese CNC with SMD-PCB using the autolevel feature, but found some weird behavior: 
    "the tip did only slightly scratch the surface"
Here's a photo of the results:
[image]
[image]

My setup:
a) tip: V-shaped 0.2mm 30degrees
b) PCB 1oz single sided, eagle design with 0.3mm minimum spacing/separation (DRC checked ok)
c) eagle's pcb-gcode (ver 3.6.2.4), with etch depth set to -0.05mm and tool to 0.2mm (I know V-bits should increase this)
d) bCNC 0.9.14-dev, with settings:
> $0=1000.000
> $1=2550
> $2=01.000
> $3=72400
> $4=04000
> $5=00
> $6=0=400.000
> $10=1400.000
> $11=0.010000
> $12=0.002.000
> $13=01500.000
> $20=01500.000
> $21=0
> $22=0
> $23=0
> $24=25.000
> $25=500.000
> $26=250
> $27=1.000
> $30=2400
> $31=4000
> $32=0
> $100=400.000
> $101=400.000
> $102=400.000
> $110=1500.000
> $111=1500.000
> $112=1500.000
> $120=100.000
> $121=100.000
> $122=100.000
> $130=125.000
> $131=85.000
> $132=35.000

e) bCNC autolevel procedure:
1. Load gcode file
2. Home gauntry, with Z adjusted to board surface: press "XYZ=0" and "Z=0" (probably not necessary). The CNC control board has a led that lights when the tip touches the surface, and manual knob control for the axis, that allows off-line adjusts
3. Probe-Autolevel, press "Clear" and "Zero" buttons
4. Probe-Autolevel, Fast Probe Feed=100, Probe Feed=10, TLO=0, Probe Cmd="G38.2"
5. X, Y= 0:20mm with 5mm step, Z=-1:1
6. Probe-Autolevel, press "Scan"
7. Once completed "Run"

After many unsuccessful test, here's some changes I did:
1. eagle's pcb-gcode etch depth set to -0.1mm, but same results
2. later I check repeatability (due to machine capabilities or PCB holding method):
- When I repeat the Autolevel procedure: Clearing previously the probe data, pressing the "Move to Origin" control button, press "Scan" and save probe data to file "XX.probe", the same exact results appear
- But if I repeat the Autolevel procedure, but manually using the knob to readjusting the Z=0, and pressing "Zero", some differences appear (max difference of 0.05mm):
[image]

I searched and read a lot, probably is my mistake, but honestly I can't find what I'm doing wrong..
Thanks in advance for your help

MARIOBASZ, 2020-06-18:
Zero button where with Z probe Z = 0?
You can always select all the gcode and make a cut 0.05mm deeper than the original

Harvie, 2020-06-19:
You can manualy enter Z, no need to edit g-code... For example enter Z of 0.01mm instead of 0mm to cut a bit deeper.

jorge-chavez, 2020-06-19:
Thank you for the answers
Let me explain what I've just do, if I correctly understand your tip:
After scanning the board (with the previous procedure), I manually modify WPos-Z 0.01mm less,  run the gcode and this is what I got:
[image]
The etch is so deep, around 0.5mm, and the board is totally useless

I think the problem can be due the wasteboard (sacrificial layer) I use, 3mm MDF
When I perform autolevel measures, as a minimal pressure is applied, no problem seems to happen.
But when etching, MDF behaves non-rigid, and there's a compression effect, that's why only scratch appears. 
Once the copper layer is breaks, some of the accumulated compression is released, and the bit can be much deeper than expected. 

To verify it, I'll check with a more rigid wasteboard, like a spare FR4 board over the aluminum plate. Once I have results I'll share

MARIOBASZ, 2020-06-19:
what bits are you using slower speeds?
use F 500 or less (in mm), and very low values to protect the fragile tip. Siva my work was satisfactory, it should be for you.
Each machine is different, will everything be rigid there?
If one pass just skims the surface, I would go deeper into passes of -0.05 to find out how to "trick" the machine next time.
FR4 decidedly good choice.
Good luck and hugs.

jorge-chavez, 2020-06-19:
The bit I'm using is a V-shape 30º 0.2mm. Once I'll receive, I'll perform tests with V-shape 10º 0.1mm, finer but weaker/short life...
The spindle is based on a 775 24v motor, so speeds are not so high (max 8300rpm/min). I use 7300 rev/min, max efficiency point
XY Feed rates are 200mm/min and Z Feed rate 100mm/min, perhaps too slow¿?
The offset you told me also applied with TLO field, should this field an alternative to an uniform deeper drill? The sign of TLO had very weird effects to me...
Thanks again for your help

jorge-chavez, 2020-06-19:
One side related question, concerning the "Scan" pink squared button in the Autolevel tab, not the "Scan" gear button. The cursor tips says "Scan autolevel margins", what is the purpose of this button?
I performed an autolevel procedure in an area, lets say 20x20mm, that matched design perimeter.
Then I press "Scan" button (pink square in the Autolevel tab), I assumed than the bit will travel along the perimeter with "corrected Z=0", but it didn't, it travelled with "non-corrected Z=0".
I thought that once the autolevel procedure was performed, the Z=0 plane was always the corrected one
That's why if the gcode design was loaded, no further actions were needed (i.e. no need to press the "Autolevel" button), the coordinate system is changed with interpolated probed data

Perhaps I'm a bit confused or I missed some detail