vlachoudis/bCNC #1430: %wait seems not to work

rgodin974, 2020-06-19:
Hello, 
It seems to have an issue with the %wait command.
I'm trying to run the code below :

`%msg Probing circle...
G92.1
%wait
G38.2 X-50 F50
%wait
%x1 = float(wx);
%y1 = float(wy);
%print(x1);
%print(y1);`

I want to display the results of the probing (wx and wy). But bCNC runs all the code without any pause. %wait command seems not to work.
Results are displayed before the end of probing even if the %wait command is used.
Therefore the wx and wy values are false.
What's wrong : my code or a bCNC issue ?
Thank you for your help.

rasciodc, 2021-05-30:
Dude, I've tried to use it before but got stuck on this 'wait' command, see issue #1289 and #1304. Looks to me like a dead end... I don't really think there is actually anyone looking into that. I'd do it if I programmed python.

If you find any alternatives to this, please let me know. This would really change things in my workflow!

MARIOBASZ, 2021-05-30:
I am clumsy with python. And bad memory. I ran into this problem some time ago (work still to be concluded: low-cost automatic tool changer, the movements were carried out satisfactorily), if I remember correctly the problem is in updating the values of the variables, check the ProbePage.py page And in CNC.py how they solved this, see how it solves it to perform probe or tool change (here is a line "remember position")