vlachoudis/bCNC #1432: Cannot install on raspberry pi

lakewaydave, 2020-07-06:
Tried installing on raspberry pi 3 but keep getting errors. I upgraded from Python 2.7.9 to 2.7.16 but I keep getting the same errors when trying to install.

I am using this command to install: sudo python2 -m pip install bCNC or sudo python2 -m pip install --upgrade bCNC

See the error message below. I get this regardless of using Python 2.7.9 or 2.7.16 .

┌─ 11:34 bash:raspberrypi  ~
└> sudo python2 -m pip install --upgrade bCNC
Downloading/unpacking bCNC
  Downloading bCNC-0.9.14.306.tar.gz (1.3MB): 1.3MB downloaded
  Running setup.py (path:/tmp/pip-build-CcF0zP/bCNC/setup.py) egg_info for package bCNC
    /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'long_description_content_type'
      warnings.warn(msg)
    Running bCNC setup...
    error in bCNC setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers
    Complete output from command python setup.py egg_info:
    /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'long_description_content_type'

  warnings.warn(msg)

Running bCNC setup...

error in bCNC setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip-build-CcF0zP/bCNC
Storing debug log for failure in /root/.pip/pip.log

lotrias, 2020-07-09:
I was having the same issues trying to install on Python 2.7.6 on my ARM based tablet and was able to get past that error message once I installed setuptools
`pip2 install --upgrade setuptools`

Also make sure you have installed the following as noted on the bottom on the installation wiki for ARM systems like the Raspberry Pi
`apt-get install python-pip python-setuptools python-dev python-tk libjpeg-dev zlib1g zlib1g-dev`

Harvie, 2020-07-09:
I don't know. people complain all the time about installing on raspberry. i've recently buyed my first raspberry, installed raspbian, did pip install bCNC and everything worked first try... i don't get why people have all these troubles...