vlachoudis/bCNC #1435: GRBL 1.1 ?

phil123456, 2020-07-24:
Hello,

I just installed a GRBL 1.1 Board (https://www.aliexpress.com/item/4000163253874.html?spm=a2g0s.9042311.0.0.4e324c4dWp6Cb9)

but the spindle does not run (it works with the control board so it's not a hardware issue)

is bCNC not compatible with GRBL 1.1 ?

the xyz axes work fine

regards

Harvie, 2020-07-24:
Hello, please flash your own GRBL firmware into the board. If you buy chinese pre-flashed GRBL boards, they usualy have some kind of bulls*t modified firmware in them. Also they often report incorrect version.

phil123456, 2020-07-24:
it works nice with Candle

Harvie, 2020-07-24:
To be honest. You can't even know what firmware you run unless you've flashed it yourself. How am i supposed to support such case?

In readme file there is guide to serial debugging https://github.com/vlachoudis/bCNC/blob/master/README.md

You might use it to capture the communication and compare it with what candle does. Then i can fix the problem if there is some.

phil123456, 2020-07-25:
are there so many different commands to run the spindle ? I could try typing them in the Candle console, try see which one works

Harvie, 2020-07-27:
So you can start spindle using bCNC button and slider GUI?

If you go to bCNC config, there is "header gcode" and "footer gcode". It is executed before g-code file. By default it should contain code to turn on spindle and wait a bit before executing code.