vlachoudis/bCNC #1436: Autolevel problem

ours-polaire, 2020-07-26:
Hello,
I have the following problem with autolevel: after updating gcode with the current scanned Z levels there are uncut sections. If I set a new value on Z FlatCAM there is no improvement. Autolevel is behaving inappropriately. For example, if a section is more elevated there, the autolevel raises the level too much and makes its existence meaningless. Does anyone know how to solve the problem?
I use version 0.9.14 and Ubuntu 20.04.

Thanks!

Harvie, 2020-07-27:
How exactly you used the autolevel... Have you set the XY zero for autolevel?

[image]

ours-polaire, 2020-07-28:
Hello,
Thank you for the remark.
It seems that here in my problem. Can you give me instructions on how to use it because I have trouble understanding the other explanations for the button.

carl1961, 2020-07-28:
Watch this video https://www.youtube.com/watch?v=3wBwjFgCjg4&t=326s