vlachoudis/bCNC #1438: Autolevel hangs

jeskata, 2020-08-02:
I can do a dry run of the code and I can also probe, but when I start autolevel (which has worked before with an old version) it seems to just hang after moving to the first scan point.

Here is the terminal output:
[image]

And my settings:
$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=1 (status report mask:00000001)
$11=1.000 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$14=1 (auto start, bool)
$20=1 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=50.000 (homing feed, mm/min)
$25=635.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=0.000 (homing pull-off, mm)
$100=5120.000 (x, step/mm)
$101=5120.000 (y, step/mm)
$102=3600.000 (z, step/mm)
$110=300.000 (x max rate, mm/min)
$111=300.000 (y max rate, mm/min)
$112=150.000 (z max rate, mm/min)
$120=16.000 (x accel, mm/sec^2)
$121=16.000 (y accel, mm/sec^2)
$122=16.000 (z accel, mm/sec^2)
$130=168.000 (x max travel, mm)
$131=110.000 (y max travel, mm)
$132=50.000 (z max travel, mm)

I am using https://github.com/CarlosGS/grblForCyclone , and have tried to clear the EEPROM and reinstalling GRBL.
Have tried connecting as either GRBL0 or GRBL1.

Harvie, 2020-08-02:
Are there any error messages in the console? (not the one in bCNC window, but white on black commandline that you can use to launch bcnc)

jeskata, 2020-08-02:
It keeps rolling 
000172.243 RX   0000  0D                                                .                
000172.243 RX   0000  0A                                                .                
000172.443 TX   0000  3F                                                ?               
000172.447 RX   0000  3C                                                <               
000172.447 RX   0000  49                                                I                
000172.447 RX   0000  64                                                d                
000172.447 RX   0000  6C                                                l                
000172.447 RX   0000  65                                                e                
000172.448 RX   0000  2C                                                ,                
000172.448 RX   0000  4D                                                M                
000172.448 RX   0000  50                                                P                
000172.448 RX   0000  6F                                                o                
000172.448 RX   0000  73                                                s                
000172.448 RX   0000  3A                                                :                
000172.448 RX   0000  31                                                1                
000172.448 RX   0000  38                                                8                
000172.448 RX   0000  2E                                                .                
000172.448 RX   0000  32                                                2                
000172.448 RX   0000  37                                                7                
000172.448 RX   0000  35                                                5                
000172.449 RX   0000  2C                                                ,                
000172.449 RX   0000  31                                                1                
000172.449 RX   0000  38                                                8                
000172.449 RX   0000  2E                                                .                
000172.449 RX   0000  31                                                1                
000172.449 RX   0000  30                                                0                
000172.449 RX   0000  30                                                0                
000172.449 RX   0000  2C                                                ,                
000172.449 RX   0000  2D                                                -                
000172.449 RX   0000  35                                                5                
000172.449 RX   0000  2E                                                .                
000172.449 RX   0000  30                                                0                
000172.449 RX   0000  30                                                0                
000172.450 RX   0000  30                                                0                
000172.450 RX   0000  3E                                                >                
000172.450 RX   0000  0D                                                .                
000172.450 RX   0000  0A                                                .                
000172.650 TX   0000  3F                                                ?               
000172.654 RX   0000  3C                                                <               
000172.654 RX   0000  49                                                I                
000172.654 RX   0000  64                                                d                
000172.654 RX   0000  6C                                                l                
000172.655 RX   0000  65                                                e                
000172.655 RX   0000  2C                                                ,                
000172.655 RX   0000  4D                                                M                
000172.655 RX   0000  50                                                P                
000172.655 RX   0000  6F                                                o                
000172.655 RX   0000  73                                                s                
000172.655 RX   0000  3A                                                :                
000172.655 RX   0000  31                                                1                
000172.655 RX   0000  38                                                8                
000172.655 RX   0000  2E                                                .                
000172.655 RX   0000  32                                                2                
000172.655 RX   0000  37                                                7                
000172.656 RX   0000  35                                                5                
000172.656 RX   0000  2C                                                ,                
000172.656 RX   0000  31                                                1                
000172.656 RX   0000  38                                                8                
000172.656 RX   0000  2E                                                .                
000172.656 RX   0000  31                                                1                
[... cut ...]
000172.863 RX   0000  30                                                0               
000172.864 RX   0000  30                                                0               
000172.864 RX   0000  30                                                0               
000172.864 RX   0000  3E                                                >               
000172.864 RX   0000  0D                                                .               
000172.864 RX   0000  0A                                                .

I am using KUbuntu 20.04.

Harvie, 2020-08-02:
So you have serial spy enabled? What if you disable it?

jeskata, 2020-08-02:
Ah, it might be I selected that for serial port. I will continue next time I go to the workshop, in a few days.

jeskata, 2020-08-08:
This is all it prints: 

new-config Utils <configparser.ConfigParser object at 0x7f2088cd0730>
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
self.canvas.winfo_id(): 60817493
Not Loading 6 axis displays
Not Loading 6 axis displays
Will clean after this operation

jeskata, 2020-08-09:
Also, Status always shows "Not connected" even though it clearly communicates.
Manual control of spindle has no effect even though it starts when I do a run of the gcode.

Mungbeanz, 2022-10-02:
I have this same issue.  What information do you want from me to pinpoint the cause.  I take it we didnt find a solution yet?