vlachoudis/bCNC #1439: Tabs on Profiles

rainerschulte, 2020-08-03:
Hi Folks!

I have a problem creating Tabs on Parts with Profiles. Maybe my workflow is wrong... But what is it?

I have a rectangular with a circle inside. For the outer contour I chose Profile outside, then Cut. When adding Tabs The Tabs are shown on the top (0 mm), Material is 3 mm thick and ends in -3 mm. The Tabs are created not on the Profile line, but on the original part edge. 

Kind Regards
Rainer

Harvie, 2020-08-04:
When creating the tabs, select the profile line instead of the original shape :-)

Workflow is like this:
Profile -> Tabs -> Cut

rainerschulte, 2020-08-04:
This is they decribed way... [https://vimeo.com/444547780](url) unfortunately not working

Harvie, 2020-08-04:
Can you send me the file that are you using and your bCNC.ini?

Harvie, 2020-08-04:
What if you uncheck the "only leave selected islands uncut" in the cutting dialog???

rainerschulte, 2020-08-04:
Hi! I found a solution. Once the Profile is created and the Pads then, in Editor I have to highlight both and cut then. 
The result now is the attached gcode-File. 
[bcnc.zip](https://github.com/vlachoudis/bCNC/files/5023643/bcnc.zip)

Harvie, 2020-08-05:
I told you. You have "only leave selected islands uncut" enabled. That means it will only respect islands/tabs that are highlighted.

closed 2020-08-05 by Harvie