vlachoudis/bCNC #1444: python3 unicode is not defined exception

Seikilos, 2020-08-16:
Hi,
I am on Ubuntu 20.04.1 LTS and installed bCNC 0.9.14-dev (.306) version.
Running it with python 3 produces unicode errors:

> Traceback (most recent call last):
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/Utils.py", line 464, in __call__
>     return self.func(*args)
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/lib/tkExtra.py", line 1222, in handleKey
>     self._items.append(unicode(item))
> NameError: name 'unicode' is not defined
> Traceback (most recent call last):
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/Utils.py", line 464, in __call__
>     return self.func(*args)
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/lib/tkExtra.py", line 1222, in handleKey
>     self._items.append(unicode(item))
> NameError: name 'unicode' is not defined
> Traceback (most recent call last):
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/Utils.py", line 464, in __call__
>     return self.func(*args)
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/lib/tkExtra.py", line 1222, in handleKey
>     self._items.append(unicode(item))
> NameError: name 'unicode' is not defined
> Traceback (most recent call last):
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/Utils.py", line 464, in __call__
>     return self.func(*args)
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/lib/tkExtra.py", line 1222, in handleKey
>     self._items.append(unicode(item))
> NameError: name 'unicode' is not defined
> Traceback (most recent call last):
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/Utils.py", line 464, in __call__
>     return self.func(*args)
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/lib/tkExtra.py", line 1222, in handleKey
>     self._items.append(unicode(item))
> NameError: name 'unicode' is not defined
> Traceback (most recent call last):
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/Utils.py", line 464, in __call__
>     return self.func(*args)
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/lib/tkExtra.py", line 1222, in handleKey
>     self._items.append(unicode(item))
> NameError: name 'unicode' is not defined
> Traceback (most recent call last):
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/Utils.py", line 464, in __call__
>     return self.func(*args)
>   File "/home/samuel/.local/lib/python3.8/site-packages/bCNC/lib/tkExtra.py", line 1222, in handleKey
>     self._items.append(unicode(item))
> NameError: name 'unicode' is not defined

Seems like python3 has some breaking changes to 2 regarding unicode / str functions. See https://stackoverflow.com/questions/19877306/nameerror-global-name-unicode-is-not-defined-in-python-3

Harvie, 2021-01-07:
Can you please check latest version? it should be fixed...

Seikilos, 2021-01-08:
Version 0.9.14.**312** fixed the issue. Great Job. Thank you! :)

closed 2021-01-08 by Harvie