vlachoudis/bCNC #1449: application hangs after partial startup

jeaton2411, 2020-08-26:
After several week of problem free use, bCNC stopped loading after a close / restart to save some acceleration configuration settings.

When I launch the application (from desktop file or command line) the application loads about halfway and hangs.  Have to reboot to terminate.

Looks like this:  https://photos.app.goo.gl/phtNrvixgY91twCc6

Have attempted update/upgrade of system and update of bCNC.

Help?  Was planning a live video demonstration for Robotics students this evening.

pmpeter1, 2020-09-20:
I've seen the same problem using Ubuntu 18.04 when running with both python2 and python3.

The workaround that seems to help us to run bCNC under strace.

% strace python -m bCNC

This seems to change the timing enough that it avoids the bug and runs normally.

Harvie, 2020-09-21:
Can you delete .bCNC configuration file in your home directory?

pmpeter1, 2020-09-22:
Same behavior after removing the .bCNC config file.   I saved the old copy and diff'ed it with the new copy and they are identical (except for file history and window size).   This was a clean install after rebuilding the OS, so I never had other/older copies of bCNC installed.

jeaton2411, 2020-09-22:
Just want to add that sudo strace allows it to open on my Raspberry Pi
install as well.