vlachoudis/bCNC #1451: UI icons look weird

EricPobot, 2020-08-30:
Hi,

I'm pretty sure the problem isn't located in bCNC itself, but I can't find where and how to fix it. 

All the icons of the UI are displayed as if there were scaled up. This results in a not quite nice looking UI, as visible on this partial screenshot:

[image]

What can cause this? Could it be related to fonts, text size settings, or some UI widget dimension setting? The icons GIFs look OK when displayed with any graphical application. 

I'm running bCNC Git version (re-pulled today and reported as 0.9.14-dev) on two Linux Mint 19.3 machines, without any special system setting WRT display resolution or alike. I use a fully isolated Python 3.8.2 virtual environment managed by pyenv to ensure there is no lib conflict with system-wide ones. The UI configuration settings are the default ones. I have noticed this problem since the first time I've used bCNC, more than a year ago now.

Apart from this, bCNC runs fine, and although this doesn't prevent from working with it, people I show my CNC to tend to focus on this detail only and react as if the software was crap. Which it is obviously not but you know the problem with the first impression ;) 

Thanks in advance for any feedback

yasapl, 2020-08-30:
Hi,

Make sure the option "Double size icon" in config is not on.

[image]

EricPobot, 2020-08-30:
@yasapl Many thanks for the reply. 

I found it by myself 5 minutes ago and came back here to auto-close the issue.

Best regards, and kudos to the bCNC team for the awesome work.

closed 2020-08-30 by EricPobot