vlachoudis/bCNC #1456: Want to edit bCNC, need some help

macieksoft, 2020-09-07:
I have secondary touchscreen on my computer. I think it would be useful to create some HMI panel for bCNC that would mimic the actual CNC control panel.

The problem is I have no experience in Python, it is quite different from VB.net I used to use. However I am ready to learn.

Is there anybody who could help? I need a way to create a new window (separate from main bCNC window) and put there some buttons and stuff. Buttons should not be the problem, but i have no idea on how to create a tkinter new window that would open when opening bCNC and run in parallel with normal bCNC window.

Can anybody make a template (.py file) that would create empty tkinter window when starting bCNC? Then I should be able to populate this window with my own controls. I want my window to be running in parallel with the normal bCNC window, to have normal bCNC window on my main screen and my window on external touchscreen. Obviously i want to be able to import the bCNC variables and use bCNC functions, so i will need to import them, i guess i know how to do that.

Harvie, 2020-09-07:
Ribbon contents can be modified in: https://github.com/vlachoudis/bCNC/blob/master/bCNC/bCNC.ini
Rest of ui is in files called `*Page.py`

macieksoft, 2020-09-07:
I know how to edit control page. I was already trying that.
The thing is that that i want to create separate window, not embedded into bCNC window. The window that i could have on separate screen. Then i could populate it with buttons that would mimic actual CNC HMI.

macieksoft, 2020-09-07:
I am using following example to open windows:

import Tkinter as tkii

class Demo1(tkii.Frame):
    def __init__(self):
        tkii.Frame.__init__(self)
        self.pack()
        self.master.title("Demo 1")
        self.button1 = tkii.Button(self, text = "Button 1", width = 25,
                               command = self.new_window)
        self.button1.grid(row = 0, column = 1, columnspan = 2, sticky = tkii.W+tkii.E+tkii.N+tkii.S)

    def new_window(self):
        self.newWindow = Demo2()

class Demo2(tkii.Frame):
    def __init__(self):
        new = tkii.Frame.__init__(self)
        new = tkii.Toplevel(self)
        new.title("Demo 2")
        new.button = tkii.Button(text = "Button 2", width = 25,
                               command = self.close_window)
        new.button.pack()

    def close_window(self):
        self.destroy()

Saved the file as maciek.py

Now how can i run this example from inside of bcnc __main__.py?
I know i have to import it with something like from maciek import *
I still have no idea on how to make this example to be executed from inside __main__.py, i want it to open those example windows defined in maciek.py file together with normal bCNC window, so i can have 2 windows, the normal bCNC window and my separate window.

macieksoft, 2020-09-09:
I am finally able to open my separate window.
I was experiencing numerous indent problems thanks to notepad++ that was inserting bunch of spaces whenever i pressed TAB button. Hope that now i will be able to get the thing i want.

Thank you anyway bCNC team.

macieksoft, 2020-09-10:
First tests of HMI GUI, very incomplete for now. Just some blinkenlights and trouchscreen buttons for fun.
First row: connection, IDLE light, start, feed hold, abort, unlock, reset.
Second row: spindle stopped light, spindle on full RPM light, manual spindle ON/OFF button, mr. snail (he sets the feed override to 25%), mr. rabbit (lights up when feed override is >100%)
[image]

macieksoft, 2020-09-24:
Progress on the panel is going on, here you can see it:
https://www.youtube.com/watch?v=cyALOzflVuQ&feature=youtu.be

I have buttons for various stuff that i use all the time, especially the jog steps (separate for XY and Z axis), work offset zero setting, Z probing, moving in Z axis to safe Z height, feed and rapid overrides, mousewheel JOG axis select (can be also done using programmable buttons on my Logitech G600 mouse) and Z<0 interlock (prevents going into the material in Z axis, at least in theory and when you don't have jog runaway).

Harvie, 2020-09-24:
bCNC has web pendant communicating with bCNC over HTTP API... Why don't you use that HTTP API to control bCNC from your custom application? It is even easy to add more API features if you need it.

macieksoft, 2020-09-25:
I will probably use HTTP API for making STM-32 pendant in future, that would be nice thing.
Instead of using external app I just edited some of the bCNC code, added new tkinter window with all the buttons and stuff, that was the simplest way of doing that.