vlachoudis/bCNC #1458: Keyboard autorepeat causing runaway jog (busting the limits/endstops)

macieksoft, 2020-09-14:
If you hold down a keyboard key and autorepeat feature in Windows is enabled (almost everyone has it enabled, this is Windows default behavior) you may get runaway jog and hit the limits, that especially happens when using larger jog step settings like 5 or 10mm per press.

This is caused by the way Windows (and other systems too, that's a known thing) handles keyboard. When you hold the keyboard key, autorepeat function kicks in and system is seeing that like you was quickly pressing that key over and over. That means sending whole lot of jog commands to GRBL causing a runaway jog and hitting the limits.

The solution would be to disable the autorepeat feature when using bCNC (and restore the default when closing bCNC). I have no idea how to do that from Python, this is system specific by the way. In Windows you can disable this in control panel: https://answers.microsoft.com/en-us/windows/forum/windows_xp-performance/can-you-disable-the-keyboard-autorepeat-function/ce1b29bc-544b-e011-8dfc-68b599b31bf5

Harvie, 2020-09-15:
This is a known bug. Just configure bCNC to use step shorter than the autorepeat interval of your keyboard.

macieksoft, 2020-09-15:
I am using 3018PRO mini CNC from Aliexpress and for me, with default settings of GRBL controller safe step in XY axis is 0.6 and safe step in Z axis is 0.3.

As safe step i mean setting that doesn't causes runaway jog, so when i let go of the keyboard button jog stops right away.
In my experimental version i have touchscreen button that sets 0.6 step on XY and 0.3 on Z, that ensures that when i release keyboard button the jog stop right away.

Now here is the trick, how to make this automatic? I mean making bCNC automatically set safe step when it detects keyboard repeat. Some kind of timer would be needed, if it detects that intreval between key presses is very short (when the repeat kicks in) and step settings are higher than safe ones it could set the safe step values to make it prevent runaway jog. I have no idea on how to implement timer in Python. I need a timer that counts time between function calls.

Another nice feature (however not directly related) that i have just implemented is mousewheel stepping, to use mouse as MPG, that is so useful i think. Maybe worth to implement in official version of bCNC?

Harvie, 2020-09-16:
Actualy GRBL does have new jog protocol, which is currently only partialy implemented. It can be used for expected kind of joging, eg.: "jog only while i hold the button and then stop immediately when button is released"...

macieksoft, 2020-09-16:
Joystick jog would be nice with ESP-32 over wifi. It would be easy to connect those little joystick modules and some encoders. That could make hybrid (joystick + encoder) MPG. bCNC already had pendant feature, so it should be possible to use ESP-32 with it.

Harvie, 2020-09-16:
Isn't there some kind of HTML5 joystick for smartphones available in bCNC web pendant? I remember reviewing such code, but i can't recall if we ended up merging it...

LYHTSPD, 2020-09-16:
I would still love to figure out how to make this work with bCNC.
http://www.machinekit.io/docs/drivers/xhc-whb04b-6/

Harvie, 2020-09-16:
> http://www.machinekit.io/docs/drivers/xhc-whb04b-6/

This is too specific to fit in bCNC's scope of interrest, however it should be quite easy to take whatever userspace device driver you can get (even the one from LinuxCNC project) and modify it to send commands to web API of bCNC. If you miss some features of the API, we can change that.

LYHTSPD, 2020-09-16:
That is what I have been looking at, but I am still a novice at linux, so I
have not gotten very far with it.

-Brandon Dilger