vlachoudis/bCNC #1462: Arrow Buttons vs. Command string

wbatie, 2020-09-18:
I am stuck and this is the only site I found that talks about the bCNC software.
I am using bCNC on a Raspberry Pi 3B+ with a mill conversion to CNC. I am using a Teensy 4.1 on a Phil Barrett board for the controller.
I have a G1 X positive number command and the Right Arrow Button moving the table in the same direction (to the left), and the opposite is true.  The Down Arrow Button and the G1 Y positive number command moves the table in opposite direction. No matter what I try: changing $3, adding ’swap’ to .bCNC and setting it to non-zero, reversing stepper leads on one coil, all results movement in opposite direction.
I would like the  Y-axis Positive number command and the Down Arrow Button to move in the same direction and to the front. 
I think the solution is to change code that the two Arrow Buttons send. I have change many parameter such as BLACK_DOWN and _UP, moveYup, moveYdown. Tried finding where cmd and ystring is set for Arrow Buttons.
So my question is; what is the .py file where pressing the Arrow Button gives the string “G0 Y+ or G0 Y-. I assume just change these would do what I want. 
Another question is if I change a .py file is there anything else I need to do?

Harvie, 2020-09-18:
> Teensy 4.1

What firmware? I am not aware about any firmware for teensy being officialy supported by bCNC...

Harvie, 2020-09-18:
If your machine is moving in oposite directions than buttons that you press, simply reverse the stepper motor cabling, this will do the trick. Another possibility is to configure this in your firmware if it is supported.

https://www.youtube.com/watch?v=-iubywKkNEE
https://electronics.stackexchange.com/questions/70643/how-to-reverse-rotation-direction-of-stepper-motor

wbatie, 2020-09-18:
The firmware is grblHAL.
Reversing wire, #3 or swap as I said doesn't solve my problem. Yes the all reverse direction, but the Y Arrow Button and G1 Y commands are opposite. I would like the to be in the same direction and there not. So, changing the commands the Arrow Button sends seam to be the only solution. Just wanted to know what .py file the come from.

closed 2020-09-18 by wbatie

Harvie, 2020-09-18:
Please delete this mess. Ok. grblHAL is certainly firmware that we want to support.
Can you please make me a picture showing the problem? Usualy CNC machines are set up like this:

X+ right
Y+ forward (away from user)
Z+ up (to the ceiling)

[image]

wbatie, 2020-09-18:
After thinking about this in graphical terms, there is no problem. Some how I had UGS working with the Down Direction being the same as Y+, thus the confusion.
I like bCNC much better the UGS. My only complaint is being an old person it would be nice if things were a little bit bigger.
Thank you for your help in clearing this up.

Harvie, 2020-09-19:
>  My only complaint is being an old person it would be nice if things were a little bit bigger.

There is config option that makes all icons twice the size, if you have sufficiently big screen, you can use it. It's primarily meant to be used with touchscreens, so icons are easier to press using fingers. I think it makes stuff less visualy appealing, because the icons are bitmap, but it might work for you.