vlachoudis/bCNC #1464: bCNC freezes part way into g code Errno 22 Errno 5 GRBL

lotrias, 2020-09-25:
So I'm running the most recent version of bCNC on a linux tablet and on almost every program I run, it halts midway through producing either Errno 5 or Errno 22. I have to disconnect and reconnect to my machine. I have tried a double shielded USB cable with ferrite cores. 

I don't think its on the CNC's end because I've been using Universal G Code sender on my windows laptop with it for months. 

Here's the error report:
`?<Run|MPos:-74.675,38.412,0.838|Bf:14,11|FS:267,1000>
?Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/Sender.py", line 764, in serialIO
    if self.serial.inWaiting() or tosend is None:
  File "/usr/local/lib/python2.7/dist-packages/serial/serialutil.py", line 590, in inWaiting
    return self.in_waiting
  File "/usr/local/lib/python2.7/dist-packages/serial/urlhandler/protocol_spy.py", line 222, in in_waiting
    n = super(Serial, self).in_waiting
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 467, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
IOError: [Errno 22] Invalid argument`

lotrias, 2020-11-07:
I fixed the issue. It was a USB timeout issue with Linux. For anyone else with this issue, the fix is to add a udev rule so that you add a static name to the USB cable and then within bCNC, you connect to that name instead of the standard tty*

closed 2020-11-07 by lotrias

GitHubCaps, 2021-01-13:
Note to self: This should be added to the WIKI for others having this issue! Even if this is closed, others can benefit from this.

GitHubCaps, 2021-01-13:
I suppose, some of us in the *nix world already know or have it automatically implemented and overlooked these fine details.

TODO: Add instructions to do so, for those that this is not transparent