vlachoudis/bCNC #1466: After update, getting Python error on connect

jeskata, 2020-09-25:
Update to latest bCNC and now I get this when trying to open serial connection:

new-config Utils <configparser.ConfigParser object at 0x7f42d26bc6d0>
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
self.canvas.winfo_id(): 4194389
Not Loading 6 axis displays
Not Loading 6 axis displays
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/jskata/.local/lib/python3.8/site-packages/bCNC/Sender.py", line 777, in serialIO
    elif self.mcontrol.parseLine(line, cline, sline):
  File "/home/jskata/.local/lib/python3.8/site-packages/bCNC/controllers/_GenericController.py", line 197, in parseLine
    self.parseBracketAngle(line, cline)
  File "/home/jskata/.local/lib/python3.8/site-packages/bCNC/controllers/GRBL0.py", line 27, in parseBracketAngle
    CNC.vars["wx"] = float(pat.group(5))
TypeError: float() argument must be a string or a number, not 'NoneType'

This is on KUbuntu 20.04.

virkotho, 2020-10-01:
I got the same Issue with the GRBL0.py", line 27

bCNC Version: git-master  76d9782c82f947f30265be788e4f1af6911e894c

Python 2.1.17 on Ubuntu 18.04

I run a pretty old Grbl on an Arduino Uno:
$I
[0.9j.20160726:]
ok

I looked at the code and found an issue with the REGEXP from _GenericController.py:11

STATUSPAT = re.compile(r"^<(\w*?),MPos:([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*)(?:,([+\-]?\d*\.\d*))?(?:,([+\-]?\d*\.\d*))?(?:,([+\-]?\d*\.\d*))?,WPos:([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*)(?:,([+\-]?\d*\.\d*))?(?:,([+\-]?\d*\.\d*))?(?:,([+\-]?\d*\.\d*))?(?:,.*)?>$")

This is used to parse the line "<Idle,MPos:0.000,0.000,4.000,WPos:0.000,0.000,0.000>". In GRBL0.py line 27, wx,wy,wz is expected to be the 5,6,7th match. The Wpos values are - with this REGEXP at position 8,9,10.
I changed line 27-29:
			CNC.vars["wx"] = float(pat.group(8))
			CNC.vars["wy"] = float(pat.group(9))
			CNC.vars["wz"] = float(pat.group(10))
this works for me, but I have no clue if
1. the REGEXP is broken or
2. REGEXP was indentationally changed, and GRBL0.py was not updated...
3. or mybe the fix 5abe8cb98603f84ac57fdd67307e18c2b611796a needs another fix?

jeskata, 2020-10-23:
OK, I am using GRBLForCyclone (on Mega), https://github.com/CarlosGS/grblForCyclone , which is pretty old.
Loaded the recent GRBL on an Arduino Uno and that does seem to communicate properly.

bosd, 2020-11-16:
Glad to see the problem is solved. @Harvie 
This one can be closed