vlachoudis/bCNC #1467: SVG lift to safe Z for G0 moves

spanner888, 2020-09-26:
SVG import works wonderfully, but for a CNC Mill, it would be fantastic to lift the tool to safe Z for G0 moves.

I have tested some code with hard coded heights as shown below, in file bCNC/lib/svgcode.py, but am at a loss as to best way to give user options for safe Z and the cutting depth. Can you give me some pointers, or I am just as happy if you can complete the work.

@@ -32,7 +32,9 @@ class SVGcode:
 			subdiv=max(1,round(segmen
 			if isinstance(segment, Move):
+				gcode.append('G0 Z5')
 				gcode.append('G0 X%s Y%s' % (rv(segment.end.x), rv(-segment.end.y)))
+				gcode.append('G0 Z0')

Harvie, 2020-09-28:
You are supposed to execute the "CUT" command on the design.

spanner888, 2020-09-28:
My apologies, I keep forgetting bCNC works this way.

MARIOBASZ, 2020-10-05:
you mean something like this?

block.append(CNC.zsafe())
?

spanner888, 2020-10-05:
@MARIOBASZ that is the sort of thing I originally had in mind, but as @Harvie noted above, I really should use bCNC as intended and cut the produced SVG gcode from the workpiece, and no code changes are required as well.