vlachoudis/bCNC #1468: pip install fails

kaipyroami, 2020-09-29:
Using preferred pip install method fails. I think I installed all dependencies but it is possible that I am missing one.

`pip2 install --upgrade bCNC
`

```Installing build dependencies ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python /usr/local/lib/python2.7/dist-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpOYEQQe
       cwd: /tmp/pip-install-lm66Io/opencv-python
  Complete output (22 lines):
  Traceback (most recent call last):
    File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
      main()
    File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 114, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/usr/local/lib/python2.7/dist-packages/setuptools/build_meta.py", line 146, in get_requires_for_build_wheel
      return self._get_build_requires(config_settings, requirements=['wheel'])
    File "/usr/local/lib/python2.7/dist-packages/setuptools/build_meta.py", line 127, in _get_build_requires
      self.run_setup()
    File "/usr/local/lib/python2.7/dist-packages/setuptools/build_meta.py", line 243, in run_setup
      self).run_setup(setup_script=setup_script)
    File "/usr/local/lib/python2.7/dist-packages/setuptools/build_meta.py", line 142, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 448, in <module>
      main()
    File "setup.py", line 99, in main
      % {"ext": re.escape(sysconfig.get_config_var("EXT_SUFFIX"))}
    File "/usr/lib/python2.7/re.py", line 210, in escape
      s = list(pattern)
  TypeError: 'NoneType' object is not iterable
  ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python /usr/local/lib/python2.7/dist-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpOYEQQe Check the logs for full command output.```

cliff-pennalligen, 2020-09-30:
I don't think so.  I'm getting the same issue.  Yesterday, on Win10 I installed Python 2.7 with all options including set PATH.  Then I executed the command above.  First time I got the message  above.  When I executed it the second time, it used the cached data, which reveals that its trying to install opencv-python-4.3.0.38, which does not support Python 2.7.  I will post a work around once I figure it out.  Second error message below:

system32>python.exe -m pip install bcnc
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Collecting bcnc
  Using cached https://files.pythonhosted.org/packages/e9/de/66813ff9559b4013bc271a862f78a91d31044b6f7ddbb2022fb1716f538c/bCNC-0.9.14.307.tar.gz
Collecting numpy>=1.12 (from bcnc)
  Using cached https://files.pythonhosted.org/packages/14/ef/9f2eeb4ff0c733ad9149f17266e388c308e171fdb8c2415dbb472e2bbc0f/numpy-1.16.6-cp27-cp27m-win_amd64.whl
Collecting Pillow>=4.0 (from bcnc)
  Using cached https://files.pythonhosted.org/packages/5e/b6/8960697526a79bd1cb4520293078be3a10d725f23f20abbee298ebdeaabd/Pillow-6.2.2-cp27-cp27m-win_amd64.whl
Collecting pyserial<=3.0.1 (from bcnc)
  Using cached https://files.pythonhosted.org/packages/42/68/41bab8a5e3e50705f25358f043181242fa197f313aa0476908d9852c8dc4/pyserial-3.0.1.tar.gz
Collecting opencv-python>=2.4 (from bcnc)
  Using cached https://files.pythonhosted.org/packages/a1/d6/8422797e35f8814b1d9842530566a949d9b5850a466321a6c1d5a99055ee/opencv-python-4.3.0.38.tar.gz
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\Python27\python.exe' 'C:\Python27\lib\site-packages\pip\_vendor\pep517\_in_process.py' get_requires_for_build_wheel 'c:\users\owner\appdata\local\temp\tmpvx6k6w'
       cwd: c:\users\owner\appdata\local\temp\pip-install-3spfhd\opencv-python
  Complete output (22 lines):
  Traceback (most recent call last):
    File "C:\Python27\lib\site-packages\pip\_vendor\pep517\_in_process.py", line 207, in <module>
      main()
    File "C:\Python27\lib\site-packages\pip\_vendor\pep517\_in_process.py", line 197, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "C:\Python27\lib\site-packages\pip\_vendor\pep517\_in_process.py", line 54, in get_requires_for_build_wheel
      return hook(config_settings)
    File "c:\users\owner\appdata\local\temp\pip-build-env-zd2qhq\overlay\Lib\site-packages\setuptools\build_meta.py", line 146, in get_requires_for_build_wheel
      return self._get_build_requires(config_settings, requirements=['wheel'])
    File "c:\users\owner\appdata\local\temp\pip-build-env-zd2qhq\overlay\Lib\site-packages\setuptools\build_meta.py", line 127, in _get_build_requires
      self.run_setup()
    File "c:\users\owner\appdata\local\temp\pip-build-env-zd2qhq\overlay\Lib\site-packages\setuptools\build_meta.py", line 243, in run_setup
      self).run_setup(setup_script=setup_script)
    File "c:\users\owner\appdata\local\temp\pip-build-env-zd2qhq\overlay\Lib\site-packages\setuptools\build_meta.py", line 142, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 448, in <module>
      main()
    File "setup.py", line 99, in main
      % {"ext": re.escape(sysconfig.get_config_var("EXT_SUFFIX"))}
    File "C:\Python27\lib\re.py", line 210, in escape
      s = list(pattern)
  TypeError: 'NoneType' object is not iterable
  ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Python27\python.exe' 'C:\Python27\lib\site-packages\pip\_vendor\pep517\_in_process.py' get_requires_for_build_wheel 'c:\users\owner\appdata\local\temp\tmpvx6k6w' Check the logs for full command output.

cliff-pennalligen, 2020-09-30:
@kaipyroami so what worked for me is running this prior to the command above:
pip2 install opencv-python==4.2.0.32
then in the bcnc install it skips opencv-python

Harvie, 2020-09-30:
> pip2 install opencv-python==4.2.0.32
> then in the bcnc install it skips opencv-python

So perhaps we should put this opencv version to the setup.py?

cliff-pennalligen, 2020-09-30:
Yes I think so

kaipyroami, 2020-09-30:
> @kaipyroami so what worked for me is running this prior to the command above:
> pip2 install opencv-python==4.2.0.32
> then in the bcnc install it skips opencv-python

Boom! That's it!
Thanks for the quick resolution!

closed 2020-09-30 by kaipyroami

Harvie, 2020-09-30:
I've put opencv-python==4.2.0.32 to setup.py... Please try if that helps...

PR #1488 (2020-11-11): Fix opencv-python version, with conditional formatting to point python 2.7 to the last supported version
Supporting both python 2.7 and version 3 is hard when the opencv module dropped python 2 support.

Commits like:
https://github.com/lgv2018/bCNC/commit/fef44488ffb7dae469f5891a1b22f002d010a40b trying to fix #1468 
fixed python 2 support, but broke python 3 support as mentioned in #1417 .

Some (LTS) distro's still defaults to python 2. @Harvie wanted to keep support for python 2.

To keep supporting both python versions conditional formatting is needed.
Python 2.7 will be pointed to the las [... cut]