vlachoudis/bCNC #1471: Contributing to wiki

Harvie, 2020-10-15:
Dne 2020-10-14 22:10, Erik Hjertholm napsal:
> Hi,
> I want to contribute to the wiki of the bCNC repo but need a source of
> information that is preferably not the source code itself, do you have
> any clue where to look? I am a total noob at this and found it very
> frustrating to try to understand all the different functions and
> buttons with little to no documentation. Being a total noob I think I
> can help increase the accessibility for other beginners like me!:)
> 
> -Erik

Harvie, 2020-10-15:
Wiki should be editable for anyone with github account: https://github.com/vlachoudis/bCNC/wiki

Harvie, 2020-10-15:
Unfortunately i am not aware of any other information source than what you can find on github. However there are some youtube videos, just search bCNC on youtube. I personaly have plans to make youtube videos about recently added features, but i am bit busy, hopefully i will get to do that.

erikhjertholm, 2020-10-15:
Hi, 
So I understand that it is better to do it here as an issue instead of via mail. However, I have A LOT of questions and I'm afraid it will get messy posting all of them here. Perhaps different threads would be the best. Regardless, would you say that this repo is where I can find most active users, or is there a forum somewhere I could join?

Harvie, 2020-10-15:
This is currently only official communication channel for bCNC project.

jean-claudeF, 2021-09-26:
Hi,
I just began experimenting with bCNC and wrote my first plugin yesterday.
It generates a hexagon for milling a nut.
```python
#!/usr/bin/python
# -*- coding: ascii -*-
# $Id$

from __future__ import absolute_import
from __future__ import print_function
__author__ = "JCF"
__email__  = "jean-claude.feltes@education.lu"

from math import sin, cos, pi
from CNC import CW,CNC,Block
from ToolsPage import Plugin

#==============================================================================
# 
#==============================================================================

def calculate_coordinates(wrenchsize):
	'''return array of (x,y) tuples for hexagon'''
	a = wrenchsize/2
	r = a / (cos(pi/6))
	
	points = []

	for i in range(0,7):
		phi = i * pi /3
		
		x = r * cos(phi)
		y = r * sin(phi)
		x= round(x, 3)
		y= round(y, 3)
		points.append((x, y))
		
	return points    

#==============================================================================
# Create a simple nut
#==============================================================================
class Tool(Plugin):
	__doc__ = _("Generate a simple nut with given wrench size")

	def __init__(self, master):
		Plugin.__init__(self, master, "simpleNut")
		self.icon  = "lamp"
		self.group = "Generator"
		self.variables = [  ("name", "db", "Nut", _("Name")),
					("wrench_size", "float",  10, _("wrench size in mm")),]
	
		self.buttons.append("exe")

	# ----------------------------------------------------------------------
	def execute(self, app):
		
		# get user input and calculate points for hexagon:	
		name = self["name"]
		if not name or name =="default":
			name = "Nut"
		wrenchsize = self["wrench_size"]
		points = calculate_coordinates(wrenchsize)
				
		# init blocks of gcode:
		blocks = []
		block = Block(name)
		
		# go G0 to first point at safe height:
		block.append(CNC.zsafe())
		x1, y1 = points[0]
		block.append(CNC.grapid(x1, y1))
		block.append("(entered)")
		
		# go G1 for the other points
		for p in points[1:]:
			x, y = p
			block.append(CNC.gline(x, y))
		
		block.append("(exiting)")
		
		# append this block of gcode 
		blocks.append(block)
		active = app.activeBlock()
		if active==0:
			active=1
		app.gcode.insBlocks(active, blocks, "SimpleNut")
		app.refresh()
		app.setStatus(_("Generated: Nut"))
```
It works fine with one bug: I cannot enter a name for the block. For me it is not very important, but maybe somebody can tell me what I have done wrong.
Another thing I would like to know is if I should really uses tabs instead of 4 spaces to indent, as it is pythonic standard.