vlachoudis/bCNC #1472: Question - How to set up easy move to a 'parked' position.

micksulley, 2020-10-18:
I have repositioned my CNC and now need to move it to a pre defined park position (not home) to fold it away.  Is there a way to set up a simple command to drive it to a predefined (absolute co-ordinates) position, similar to the Home button, which I will still need?

micksulley, 2020-10-18:
Just realised I can use the 'user defined' buttons.

closed 2020-10-18 by micksulley

MARIOBASZ, 2020-10-18:
You can also edit it in the header or footer.
Note: Self-leveling is dangerous to use together with G53: by modifying the height it modifies it as if it were G54 (or the working area in execution):
suppose that autolevel indicates "+ 0.2mm";
and the original code is
.....
G1 X 0 Y10 Z-1
(go to Z machine safe)
G53 G0 Z-5

probably run
G1 X0 Y10 Z -0.8
G0 Z-4.8
Piece and end a thousand maybe damaged!