vlachoudis/bCNC #1475: Auto level in relative coordinate system G91

cliff-pennalligen, 2020-10-27:
Hi, you must be used to people saying thanks for making bCNC just before they post their issues; really yes, thankyou.  If Github had a mechanism for posting appreciation, I'm sure it'd be used for bCNC.

I made a simple Gcode program with all relative moves (as below).  When applying auto-levelling, the output is as below.  It seems the auto levelling is treating it as absolute (move X10, move X20, move X30, intending X30 but actually X60), which made the X axis travel far in the positive direction.  I guess auto levelling is not yet implemented for relative coordinate systems?  No problem, I'll switch to absolute.  Just wanted to know the state of the art.

Gcode:
G91
G1Z3F100
M3S300
G1Z-3.5F100
G1X30F1200
G1Y0.5F1200
G1X-30F1200
G1Y0.5F1200
G1X30F1200
G1Y0.5F1200
G1X-30F1200
G1Y0.5F1200
G1X30F1200
G1Y0.5F1200
G1X-30F1200
G1Y0.5F1200
G1X30F1200
G1Y0.5F1200
G1X-30F1200
G1Y0.5F1200

----------------------------------------------------------
Terminal output:
ok
G90
ok
G91
ok
G1X0Y0Z3F100
ok
M3S300
ok
G1X0Y0Z-0.5F100
ok
G1X10Y0Z-0.64F1200
ok
G1X20Y0Z-0.785
ok
G1X30Y0Z-0.965
ok
G1X30Y0.5Z-0.9719F1200
ok
G1X30Y0.5Z-0.9719F1200
ok
G1X20Y0.5Z-0.7915
ok
G1X10Y0.5Z-0.6459
....... and so on

----------------------------------------------------------
Probe file:
0 40 5
0 10 2
-10 1 150

0 0 0
10 0 -0.14
20 0 -0.285
30 0 -0.465
40 0 -0.582

0 10 -0.068
10 10 -0.258
20 10 -0.414
30 10 -0.603
40 10 -0.762

MARIOBASZ, 2020-10-27:
autolevel is very good but has some dangerous peculiarities:
1- from what you say, one is: it does not understand relative distances, it takes them as absolute. I did not know, I did not know it.
2- Refrain from using G53: interpret as a work area in use (unless it is wrong).
My job completion is usually
G0 G53 Z-5 (safe position of the machine)
With autolevel he translated it to
G0 Z-5 plus autolevel compensation. I wrecked an end mill.
3- I'm not sure, but I think there is a conflict if you want to cut outside the probing area.

cliff-pennalligen, 2020-10-27:
@MARIOBASZ thankyou for the information.  I'll keep those points in mind.  Still, it's a good feature and has on most occasions worked flawlessly for my distorted workpieces.