vlachoudis/bCNC #1476: AttributeError: 'NoneType' object has no attribute 'split'

komputronika, 2020-10-28:
I am using Ubuntu 16.04 - 64 bit.
I get this error when running bCNC.

AttributeError: 'NoneType' object has no attribute 'split'

Please help.

bosd, 2020-11-16:
Which version of bCNC are you running? 
Are you using python2 or 3?

DodoLaSaumure, 2020-12-31:
@komputronika 
Could you please provide your full error stack ? and your version of python & bCNC ?
Have you upgraded pyserial ?

ldpgh, 2021-07-17:
Have the same issue ... my setup
- Python 3.8.10 (tags/v3.8.10:3d8993a, May  3 2021, 11:48:03) [MSC v.1928 64 bit (AMD64)] on win32
- bCNC==0.9.14.317 @ pip freeze . . . . . . btw. `bCNC.__version__` is not set in the module

ldpgh, 2021-07-17:
Stacktrace
C:\Users\00000000000>python -m bCNC
new-config Utils <configparser.ConfigParser object at 0x000000000350ED90>
Using fallback Utils.comports()!
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
self.canvas.winfo_id(): 657332
Traceback (most recent call last):
  File "C:\Programme00000000000\Python\Python38-64\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Programme00000000000\Python\Python38-64\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Programme00000000000\Python\Python38-64\lib\site-packages\bCNC\__main__.py", line 2729, in <module>
    main()
  File "C:\Programme00000000000\Python\Python38-64\lib\site-packages\bCNC\__main__.py", line 2696, in main
    application = Application(tk)
  File "C:\Programme00000000000\Python\Python38-64\lib\site-packages\bCNC\__main__.py", line 197, in __init__
    page = cls(self.ribbon, self)
  File "C:\Programme00000000000\Python\Python38-64\lib\site-packages\bCNC\CNCRibbon.py", line 115, in __init__
    self.register()
  File "C:\Programme00000000000\Python\Python38-64\lib\site-packages\bCNC\FilePage.py", line 392, in register
    self._register((FileGroup,
  File "C:\Programme00000000000\Python\Python38-64\lib\site-packages\bCNC\CNCRibbon.py", line 134, in _register
    w = f(self.master._pageFrame, self.app)
  File "C:\Programme00000000000\Python\Python38-64\lib\site-packages\bCNC\FilePage.py", line 254, in __init__
    self.comportRefresh()
  File "C:\Programme00000000000\Python\Python38-64\lib\site-packages\bCNC\FilePage.py", line 348, in comportRefresh
    for hw in i[2].split(' '):
AttributeError: 'NoneType' object has no attribute 'split'

ldpgh, 2021-07-17:
Debug with pdb.stack_trace() in comportRefresh

DodoLaSaumure, 2021-07-17:
Did you upgrade pyserial?

ldpgh, 2021-07-17:
bCNC is running on the laptop. No machines are connected.
Maybe the "none" are coming from disabled bluetooth and sd-card reader. Both are showing an exlamation mark in the USB-view.

ldpgh, 2021-07-17:
pyserial==3.0.1

ldpgh, 2021-07-17:
Have wrapped it with try/except as quick work around ... program comes up

ldpgh, 2021-07-17:
(1) python -m pip install pyserial --upgrade . . . gave me pyserial 3.5 with collision to bCNC
(2) python -m pip uninstall pyserial
(3) python -m pip install pyserial==3.0.1
establishes a working setup without the try/except workaround.

serial.Serial() is now available.
self.comportsGet() returns   `<generator object comports at 0x00000000142B8740>`   compared to previous `[('com1', None, None)]`

The question regarding pySerial version numbers was helpfull, but the real root cause with old and new version number the same, is still a miracle to me.