vlachoudis/bCNC #1481: Can not start with Python3.9: NameError: name 'sys' is not defined

kvinwang, 2020-11-07:
```
$ python -m bCNC
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/kvin/codes/cnc/bCNC/bCNC/__main__.py", line 50, in <module>
    import Utils
  File "/home/kvin/codes/cnc/bCNC/bCNC/Utils.py", line 41, in <module>
    if getattr( sys, 'frozen', False ):
NameError: name 'sys' is not defined
```

closed 2020-11-07 by Harvie

GitHubCaps, 2021-01-13:
Just a note to self: `Utils.py` already and has always had `import sys`

Where did this become a problem? 

If  "anybody is out there" and knows, please point it out!

GitHubCaps, 2021-01-13:
Another note to self: I am not using `python3.9`
Did something change?

Is/was this a problem with Travis/github or merging from other sources? 

I am using 3.8 for testing (2.7, also) don't want to compile to the latest  py version (bleeding-edge)  every time there's a new release!

Side note: Releases have been more frequently lately, before version 4 (should we yield?)

There is definitely a fallacy here, if so!

Anyways, I don't have the time to track those changes. However, I know that some of these changes should not have taken place...

kvinwang, 2021-01-15:
> Just a note to self: `Utils.py` already and has always had `import sys`

`Utils.py` did not import sys before #1483.
It works under python3.8, because we have `from tkinter import *`.

cardosorapha, 2021-04-27:
I am getting this error in python 3.9.4

PR #1483 (2020-11-07): import sys
This fixes #1481