vlachoudis/bCNC #1482: ImportError: cannot import name 'gcd' from 'fractions' (/usr/local/lib/python3.9/fractions.py)

kvinwang, 2020-11-07:
```
$ python -m bCNC
new-config Utils <configparser.ConfigParser object at 0x7effa6a47400>
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
Traceback (most recent call last):
  File "/home/kvin/codes/cnc/bCNC/bCNC/ToolsPage.py", line 1098, in __init__
    exec("import %s"%(name))
  File "<string>", line 1, in <module>
  File "/home/kvin/codes/cnc/bCNC/bCNC/plugins/spirograph.py", line 20, in <module>
    from fractions import gcd
ImportError: cannot import name 'gcd' from 'fractions' (/usr/local/lib/python3.9/fractions.py)
```

closed 2020-11-07 by Harvie

GitHubCaps, 2021-01-13:
Reminder: Converted plugin `spirograph.py` does not require `import math` or `import fractions`

Add: `lcm/gcd` into `bmath`:
`from bmath import (cos, lcm, pi, sin)`

No dependency on standard libraries (math) and no py version dependency!

GitHubCaps, 2021-01-13:
Example:
`bmath.py` add:
```python
# -----------------------------------------------------------------------------
# Delta is wise, dinosaurs must die! (I hope, nofx will forgive me ;)
# -----------------------------------------------------------------------------
def gcd(a, b):
	"""Return greatest common divisor using Euclid's algorithm."""
	while b:
		a,b = b,a%b
	return a

# -----------------------------------------------------------------------------
# floor is sufficient? I want to see something/someone sticking to the ceiling,
# preferably, my closest friend, linoleum!
# -----------------------------------------------------------------------------
def lcm(a, b):
	"""Return lowest common multiple."""
	return (a*b) // gcd(a,b)

```

`spirograph.py` change:
```python
from bmath import (cos, lcm, pi, sin)
```
`spirograph.py` required `lcm`, which in turn, required `gcd`, both have been accounted for now in `bmath`

Solves all issues to date, although, there is more to the code! (comments are optional)

PR #1484 (2020-11-07): import gcd from math as a fallback
this fixes #1482