vlachoudis/bCNC #1492: 'zip' object is not subscriptable in spline.py

Git-Bruno, 2020-12-01:
Hi,
I got an error 'zip' object is not subscriptable in spline.py opening a dxf file

  File "C:\Users\Bruno\Documents\CFAO\CNC\bCNC\bCNC\lib\dxf.py", line 1325, in convert2Polylines
    entity.convert2Polyline(self.splineSegs)
  File "C:\Users\Bruno\Documents\CFAO\CNC\bCNC\bCNC\lib\dxf.py", line 549, in convert2Polyline
    xx,yy,zz = spline.spline2Polyline(xyz, int(self[71]),closed, splineSegs, knots)
  File "C:\Users\Bruno\Documents\CFAO\CNC\bCNC\bCNC\lib\spline.py", line 188, in spline2Polyline
    print("xyz[0] ", xyz[0])
TypeError: 'zip' object is not subscriptable

My skill in Python is tiny but I managed to fix the problem with the change below in file dxf.py

def convert2Polyline(self, splineSegs):
		"""Convert complex objects (SPLINE,ELLIPSE) to polylines"""
		if self.type == "SPLINE":
			# Convert to polyline
			xyz  = list(zip(self[10], self[20], self[30])) # Added list to fix 'zip object is not subscriptable in spline.py'

This is with Python 3.8.2. 
Can this change have bad side effects ? is it safe to do the same change for all zips that may exist in bCNC code ? and is there a way to have a change working regardless of the version of Python used ?
Thanks for any advices.
Bruno

DodoLaSaumure, 2020-12-31:
@Git-Bruno 
Hi
Do you still have this problem after upgrading to the last commits ? 0.9.14.312
If no, could you please close ?
Thanks forwards
Have a good day
PIerre

Git-Bruno, 2020-12-31:
Hi,
Import of the offending dxf file works fine.
Thanks for your support.
Bruno

closed 2020-12-31 by Git-Bruno