vlachoudis/bCNC #1493: Keyboard shortcuts for macros

aerogeek217, 2020-12-03:
Is there a way to assign keyboard shortcuts to the user buttons? For example, I have a laser test fire button that I would like to be able to activate with a keyboard or pendant,  while looking at the workpiece.

Harvie, 2020-12-04:
There is "shortcuts" menu in config section. I beleive you can put some g-code in ther...

aerogeek217, 2020-12-04:
Thanks, and that was basically my backup plan. Would still be nice to be able to assign keyboard shortcuts to menu items in general, or the user buttons specifically.

Harvie, 2020-12-04:
https://github.com/vlachoudis/bCNC/wiki/CommandLine

water-rhatt, 2021-03-06:
So, just to be clear. There is no CommandLine command for triggering a user button, or am I reading this wrong? I too would like to have both a button and a keyboard shortcut assigned to that button without having to duplicate the button in Shortcuts. Not that it is impossible to do, but inelegant.