vlachoudis/bCNC #1495: xy problem linuxcnc and candle grbl

tevfikAkhisar, 2020-12-08:
[image]
[image]
[image]

[ekran.txt](https://github.com/vlachoudis/bCNC/files/5659627/ekran.txt)

Harvie, 2020-12-08:
Weird

[image]

Harvie, 2020-12-08:
What is your python version? How did you installed bCNC ?

tevfikAkhisar, 2020-12-14:
linux mint problem :(        windows no problem :)  thank you

Harvie, 2020-12-14:
`python -m pip freeze` will print version numbers of all libraries, maybe you have different version of some library than on windows?