vlachoudis/bCNC #1497: G2 only works only on some ways

arnold-b, 2020-12-12:
If you use some gcode which is generated by some tools it dont work. 
Controller will stop with idle command

%
( File created using Intuwiz Software Service )
( http://www.intuwiz.com )
( More information: http://www.intuwiz.com/circle.html )
( File created:  2020-12-12  17:10:45  )

G00 Z0.5 F70
G00 X-4.5 Y0 F70
G01 Z-0.4 F50
G02 I4.5
G01 Z-0.8
G02 I4.5
G01 Z-1.2
G02 I4.5
G01 Z-1.6
G02 I4.5
G01 Z-2
G02 I4.5
G01 Z-2.4
G02 I4.5
G01 Z-2.8
G02 I4.5
G01 Z-3.2
G02 I4.5
G01 Z-3.6
G02 I4.5
G01 Z-4
G02 I4.5
G01 Z-4.4
G02 I4.5
G01 Z-4.8
G02 I4.5
G01 Z-5.2
G02 I4.5
G01 Z-5.6
G02 I4.5
G01 Z-6
G02 I4.5
G00 Z0.5 F70
G00 X0 Y0 F70
M30
%

MARIOBASZ, 2020-12-12:
can you see if there is an error message?
I think I remember that G0 must go without advance speed F, otherwise it sends an error message (but maybe it stops a few lines later)

Git-Bruno, 2020-12-12:
Hi, I had a similar case a while back , with error 33 reported by grbl though. I fixed it by setting  'Digital digits' to 4  and 'Plotting Arc accurancy' to 0.002 in bCNC along with 'Arc tolerance' set to 0.002 in grbl.
Hope this helps.

ad1976, 2021-01-11:
I just spent 3 hours thinking some python library got updated or my grbl controller got fried while trying to run a simple cicrular helical  pocket operation generated with the new bCNC. It was absolutely killing bCNC, couldnt so much as and change the viewport type. Finally I tried just running a offset flatten operation and no problem. It's by
It a while to go yet, but but I thinkmi saw am error 33  in the terminal. I'll try after the current run is done, but 8 have high hopes for your fix, seen some other posts relating grbl which would were along those lines. Thanks man. And a big big thanks to the vlachudis for his work. Anybody know what he drinks, is it ouzo?