vlachoudis/bCNC #1498: Creating tabs from islands

creat1c, 2020-12-19:
I'm trying to mill a grid of rectangular plates held together by a series of tabs. Because of the huge number of plates and the need to have precise placement of the tabs, I did not use the "tabs" feature of bCNC but instead added the tabs as rectangles into the drawing, set them to be islands and used "leave islands uncut". See attached screenshot for settings. 
I am trying to avoid the tabs in bCNC since they are not well placed (rightmost in the image), and I have to move them manually, which is not very precise and - as I am working on an rPI and it's a big project - any working on the project in bCNC is a slideshow.

My issue with the method I used (left in the image) is that I did not find a solution how to control the height of the tabs this way. The path always raises to the "Safe Z" (10) level which is quite a bit above the surface, compare this to the path when using the tabs option (center of the image) , where it passes at tab height level (-2 )

I've worked around this somewhat by first milling down to the height of the tabs without restriction, and then in a second pass, milling further down regarding the tabs. This takes a lot of time and also this results in an uneven contour of the piece at the site of the tab. 

I would prefer the path raise only to "Surface Z" (0) level instead, or even better, to some other set level as the "tabs" option does. Maybe I have overlooked something, like a way to mark objects in a similar fashion as the tabs option does? Or is it simply not possible?

OT: Great software and thanks much for writing it!

[image]

creat1c, 2020-12-19:
I've attached the drawing used in milling if it is of any help

[project.zip](https://github.com/vlachoudis/bCNC/files/5719130/project.zip)

Harvie, 2020-12-19:
The trick is in the object name:

[image]

you see the tab has the "minz" specified in the name. when it is not specified, the "safe z" is used instead.

creat1c, 2020-12-19:
Awesome! That did the trick.

Thats for the quick response and keep up the good work!

closed 2020-12-19 by creat1c

MARIOBASZ, 2020-12-20:
If "eyelash height" (absolute depth) can be modified by "eyelash thickness", it will be more practical to use as default for different cases with different depth of cut without having to modify flange height (and possible errors).