vlachoudis/bCNC #1499: DXF import fails

leventelist, 2020-12-20:
I get this when trying to import the attached DXF.

Traceback (most recent call last):
  File "/home/lev/github/bCNC/bCNC/Utils.py", line 465, in __call__
    return self.func(*args)
  File "/home/lev/github/bCNC/bCNC/__main__.py", line 2011, in loadDialog
    if filename: self.load(filename)
  File "/home/lev/github/bCNC/bCNC/__main__.py", line 2072, in load
    Sender.load(self,filename)
  File "/home/lev/github/bCNC/bCNC/Sender.py", line 410, in load
    self.gcode.importDXF(filename)
  File "/home/lev/github/bCNC/bCNC/CNC.py", line 2401, in importDXF
    dxf.sort()
  File "/home/lev/github/bCNC/bCNC/lib/dxf.py", line 1316, in sort
    block.sort()
  File "/home/lev/github/bCNC/bCNC/lib/dxf.py", line 793, in sort
    layer.sort()
  File "/home/lev/github/bCNC/bCNC/lib/dxf.py", line 736, in sort
    sx,sy = entity.start()
  File "/home/lev/github/bCNC/bCNC/lib/dxf.py", line 411, in start
    self._calcEndPoints()
  File "/home/lev/github/bCNC/bCNC/lib/dxf.py", line 447, in _calcEndPoints
    self._start = Vector(self[10][0], self[20][0])
TypeError: 'float' object is not subscriptable

[xmas_w_logo.dxf.zip](https://github.com/vlachoudis/bCNC/files/5721326/xmas_w_logo.dxf.zip)

qmaster0803, 2021-08-06:
Same problem

thinkJD, 2021-12-12:
I have the same Problem with Inkscape dxf export. AutoCAD R12 works, AutoCAD R14 fails with the same error.

commit 68555bcd (2023-03-06): Fix edge case in DXF import, closes #1499

Correctly import lists of 1 element
closed 2023-03-06 by Harvie