vlachoudis/bCNC #1501: Make something on heigh <> 0

Ogurezzz, 2020-12-24:
Hi.
I tried to slice mesh (e.g. half sphere) and make pocket operation on selected height slice (7.2mm).
As result I get good pocket. But it is on 0 height.
Also I can't find pocket plugin file to check what is inside :)
See screenshots. Solving this issue will help to make one more step to CAM.
[image]
[image]

MARIOBASZ, 2020-12-24:
Usually the surface height is used as Z = 0 and the operations are slid to negative Z. Anyway, in the Stock tab you can declare the location of the part's surface height, depth, and safe Z.
When ordering a profiling or pocket operation, the height is generated at the surface height value (in your case it is Z = 0)
Then you order it to cut to the desired height.
I hope I have understood you and this is helpful. Regards.

Ogurezzz, 2020-12-24:
Unfortunately, there is no sense is layer greater than zero or below.
 Also block of Gcode generated by pocket plugin have Z coordinate in each G1 line. This makes difficult to edit gcode (I had idea just to edit gcode in block before "(entered)" line.
As I can see this plugin takes gcode from 1 block. It can just check block height to generate new block.
Or it can generate Z coordinate move only in beginning.
This option cold become as boolean control.

PS if there is a way to move block of Gcode along Z axis - please show me. :)

[image]

MARIOBASZ, 2020-12-25:
This?
1-select block
2- write in command line
cut -1
or
cut -2 1
( feed , Zsafe, Zsurface: information from Stock selected)

Ogurezzz, 2020-12-27:
Thank you for this commands... Looks like there is much more power in bCNC :)
I need to learn more in wiki :)

closed 2020-12-27 by Ogurezzz

Harvie, 2020-12-27:
Just for the record: Any 3D operations on STL files are currently just for experiments and fun. There is currently no way to properly handle milling of 3D objects imported from 3D meshes. But you can export 2D slice geometry from the meshes and process it in the same way as you would process any other 2D drawing (eg. from DXF).

Ogurezzz, 2020-12-27:
Anyway I've got what I need :)
And with 3D slice option I can get nice result.
[image]

Harvie, 2020-12-27:
BTW there is "bowl generator" plugin in bCNC which does the exact same hemisphere thing even without need for STL.