vlachoudis/bCNC #1503: bCNC.ico not defined

Git-Bruno, 2020-12-31:
Hi,
I am getting the error lines below when I want to check the version of bCNC using menu Files->About.
An empty window is displayed. Certainly a very minor issue.

Traceback (most recent call last):
  File "C:\Users\Bruno\Documents\CFAO\CNC\bCNC-master\bCNC\Utils.py", line 526, in __call__
    return self.func(*args)
  File "C:\Users\Bruno\Documents\CFAO\CNC\bCNC-master\bCNC\__main__.py", line 747, in about
    self.iconbitmap("bCNC.ico")
  File "C:\Python38\lib\tkinter\__init__.py", line 2071, in wm_iconbitmap
    return self.tk.call('wm', 'iconbitmap', self._w, bitmap)
_tkinter.TclError: bitmap "bCNC.ico" not defined

I changed line 747 in __main__.py from 	self.iconbitmap("bCNC.ico") to 	self.iconbitmap("%s\\bCNC.ico"%(Utils.prgpath)) to clear the error.

Alternatively a workaround is to copy the missing file from bCNC/icons dir to the bCNC root dif.

The version number shown is 0.9.14. Shouldn't is be 0.9.15 ?

Thanks to all the contributors to this great softwarefor for the quality of work. 
Bruno

PS: Best wishes to all for 2021.

Harvie, 2020-12-31:
0.9.15 is not released yet. Any reasonably recent version has both bCNC and python version displayed in window title.

[image]

However you really should instal bCNC using pip and therefore you will be able to get version using `pip show bCNC`, eg.:

```
# pip show bCNC
Name: bCNC
Version: 0.9.14.309
Summary: Swiss army knife for all your CNC/g-code needs
Home-page: https://github.com/vlachoudis/bCNC
Author: Vasilis Vlachoudis
Author-email: vvlachoudis@gmail.com
License: GPLv2
Location: /usr/lib/python3.9/site-packages
Requires: numpy, Pillow, opencv-python, pyserial
Required-by:
```

Please do yourself a favor and install bCNC using pip as described in README and wiki:
https://github.com/vlachoudis/bCNC/wiki/Installation

that way you will be able to easily update to the latest reccomended build.

Git-Bruno, 2021-01-01:
Thanks Harvie.  I reinstalled using PIP/ It is certainly the simplest way to keep the install consistent and uptodate.
pip show bCNC returns Version: 0.9.14.312. 
I will dig a bit to find if bCNC code can go elsewhere than in the site-package dir to allow for running with different versions of python.
As far as bCNC.ico not defined  .. the duck is still alive even after the install with pip :-)