vlachoudis/bCNC #1508: assigning values to variables

swarfer, 2021-01-08:
I am creating macros for one of these probes
https://openbuildspartstore.com/xyz-touch-probe-plus/

I am trying to calculate the bit diameter by probing the hole.  No problem there.
but while I can get the [diameter] value that was set in the CAM|Endmill dialog it appears that no variant of
%diameter=#newdiameter
will actually set the system CNC.vars['diameter'] as intended.

What have I missed?

rschell, 2021-01-30:
You might try "%app.cnc.__setitem__("diameter",_newdiameter)"

rschell, 2021-01-30:
Here is my gcode file that is equivalent to the OpenBuilds Control routine
[ProbeXYZ.txt](https://github.com/vlachoudis/bCNC/files/5898955/ProbeXYZ.txt)

swarfer, 2021-01-31:
> 
> 
> Here is my gcode file that is equivalent to the OpenBuilds Control routine
> [ProbeXYZ.txt](https://github.com/vlachoudis/bCNC/files/5898955/ProbeXYZ.txt)

thankyou, that is nice