vlachoudis/bCNC #1509: Web Pendant, Picture flickering

Guenni75, 2021-01-08:
Happy New year.
I have a Problem with the Web Pendant.
It is the same on Windows, Linux, Android and every Browser i can use.
bCNC is installed on Lubuntu.

https://user-images.githubusercontent.com/14917841/104018878-dbe64400-51ba-11eb-83a2-b44c898ca375.mp4

GitHubCaps, 2021-01-09:
@Guenni75 Have you tried directly from bCNC first?

Make sure that you are connected to pendant and can run some code locally via bCNC through the pendant.

Personally, I stopped using Lubuntu a long time ago because of issues with camera, web, etc. I found Linux Mint XFCE versions to fit the bill a bit better, I use the 32-bit versions for my old machines and of course the 64-bit for the newer. I stay away from Cinnamon and Mate, because I prefer lightweight OSes myself.

GitHubCaps, 2021-01-09:
@Guenni75 Did some further investigating on my end and it does have an issue when opening a file in pendant. It works in Python2, however it needs patching for Python3.

I will reply back as soon as I have a patch for it.

GitHubCaps, 2021-01-10:
@Guenni75 and anyone else that would like to test the following before a commit.

Tested in python 2.7.14 (already worked there) and 3.8.2 (lots of issues there)
[Pendant.py.zip](https://github.com/vlachoudis/bCNC/files/5792214/Pendant.py.zip)

`Pendant.py` for those that don't like to download anything suspicious:
```python
# -*- coding: ascii -*-
# $Id$
#
# Author: Vasilis Vlachoudis
# Email:  vvlachoudis@gmail.com
# Date:   24-Aug-2014

from __future__ import absolute_import
from __future__ import print_function
__author__ = "Vasilis Vlachoudis"
__email__  = "Vasilis.Vlachoudis@cern.ch"

import os
import re
import sys
#import cgi
import json
#import urllib
import tempfile
import threading

# ---------------------------------------
# For debugging issues
# ---------------------------------------
import traceback
from errno import EPIPE
try:
	broken_pipe_error = BrokenPipeError  # Python3 has: BrokenPipeError
except NameError:
	broken_pipe_error = IOError
# ---------------------------------------

try:
	import urllib.parse as urlparse
	from urllib.parse import unquote
except ImportError:
	import urlparse
	from urllib import unquote as unquote

try:
	import http.server as HTTPServer
except ImportError:
	import BaseHTTPServer as HTTPServer

try:
	from PIL import Image
except ImportError:
	Image = None

from CNC import CNC
from Utils import prgpath
import Camera

HOSTNAME = "localhost"
port     = 8080

httpd    = None
webpath  = "%s/pendant"%(prgpath)
iconpath = "%s/icons/"%(prgpath)

#==============================================================================
# Simple Pendant controller for CNC
#==============================================================================
[... 257 lines cut ...]
		httpd.serve_forever()
	except:
		httpd = None

# -----------------------------------------------------------------------------
def start(app):
	global httpd

	if httpd is not None: return False
	thread = threading.Thread(target=_server, args=(app,))
	thread.start()
	return True

# -----------------------------------------------------------------------------
def stop():
	global httpd
	if httpd is None: return False
	httpd.shutdown()
	httpd = None
	if Pendant.camera: Pendant.camera.stop()
	return True

if __name__ == '__main__':
	start()
```

Guenni75, 2021-01-10:
Hi. After Updating to Ubuntu 20..., i made another try without the pendant.py above.
Browser and bCNC were on the same PC.
This is the Terminaltext:

Exception happened during processing of request from ('127.0.0.1', 46364)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 293, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 657, in __init__
    self.finish()
  File "/usr/lib/python2.7/SocketServer.py", line 716, in finish
    self.wfile.close()
  File "/usr/lib/python2.7/socket.py", line 283, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 307, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
error: [Errno 32] Broken pipe
----------------------------------------

Next i try the pendant.py above....

Guenni75, 2021-01-10:
The same problem with the new pendant.py.

Guenni75, 2021-01-10:
I installed the last bCNC with Python 2.7 on Win10 and the Pendant only shows a warning sign ;(

Guenni75, 2021-01-10:
The same with the new pendant.py ;(

GitHubCaps, 2021-01-10:
That is a common error that comes from socket (not bCNC) and can be ignored. But what about it actually working now with the patched file?

Guenni75, 2021-01-11:
As I wrote above:

> The same with the new pendant.py ;(

GitHubCaps, 2021-01-11:
Do you have any issues with serial communication between bCNC and your CNC machine? How are you connected from android to the pendant? Have you tried connecting bCNC using spy:// from the terminal to see what's going on? 

Lots of questions, but any info you can give would really be appreciated!

GitHubCaps, 2021-01-12:
@Guenni75 At first, I did see that your video shows a part in it (it only flickered).
It seems to me that something is happening at network level connection wise.
Maybe, you can check out these #1072 and #1121
Possibly @qvisionsa or @PCSDIAS may have some info that can help you resolve this issue.

Guenni75, 2021-01-14:
@GizHubCaps   Ar first, yes.... it flickers and you can see the part not even a second. That's why I opened this issue.

Guenni75, 2021-01-14:
A have one bCnc on lubuntu and one on Win10. Both with Python 2.7.
The serial communication works without any issue.
At the Linux Version i have this flickering with and without a connection to the CNC (pendant opened on the same PC and on a remote PC).
At the Win10 Version the pendant shows no flickering, but a warning sign instead of the part.
On both Versions, moving, homing and so on works fine.
I don't know how to connect bCNC with Spy.

GitHubCaps, 2021-01-15:
I noticed that the name of the file you were running is in unicode. Have you tested with an ascii formatted file and possibly using bCNC with English? May not be an issue, but wouldn't hurt to eliminate as much as possible.

As far as the `error` and `BrokenPipeError` py2/3 respectively, those can generally be ignored,(serial is negotiating connections) sometimes I have few show and other times non at all.

The flickering may be due to network connection issues (seeing that the part at least shows up) but not sure atm. Hopefully, if someone else has had this issue before, maybe we could get some more information.

In bCNC where you connect to usb `File -> Serial 
-> Port:` is where there is options for `spy:// `
Also there's info in the README and manual.

Note: I will have something up soon to fix some encoding bugs (for both) and the file open in py3
Don't know if that will help you or not, but we'll see.

GitHubCaps, 2021-01-15:
@Guenni75 Also, if you can, zip the file and upload it here (drag/drop into comment box) so I can test it directly to see if it gives me the same issue.

truglodite, 2021-02-19:
Just reporting in I am having the same issue. I'm running the latest dev on a pi3 using python3 (installed via pip yesterday). Both local browser and remote (chrome on win10) display the flickering issue. I have very little understanding to help troubleshoot any networking issues related to this, but I have seen in chrome Dev tools the same thing as pcsdias here:
https://github.com/vlachoudis/bCNC/issues/1121#issuecomment-452682544 

The canvas keeps failing to load, and I get the same error messages on the terminal of "An established connection was aborted by the software on the host computer". There is also a repeating message in chrome debug console `Failed to load resource: net::ERR_CONTENT_LENGTH_MISMATCH`

I can share additional files/info if it would be helpful to fix this. I can say I have tried with several different gcode files and encoding does not appear to solve this. So I don't think it's anything to do with that. Also note I previously had installed bCNC with python2 a few months ago, and it was behaving the same way.

Guenni75, 2021-10-06:
@GitHubCaps It doesn't matter which file i use. It's always the same.
Since a few months, i only get a warning sign.

jsiddall, 2022-02-22:
I was having this problem with earlier builds but with the latest version it is working again, even with python3

luckypickle1, 2024-06-24:
Make sure to install all dependecies including numpy for the latest version and you should have no problems