vlachoudis/bCNC #1510: bCNC 0.9.14-dev

Barry123571, 2021-01-09:
I am brand new to all of this. Linux Github  milling. So please excuse me if this is the wrong place to register a problem
Having installed and tested up to getting the machine to respond in the correct direction I have attempted to import gcode from several sources. This results in the editor displaying the code but no image is apparent.
I am now at a loss as to how to proceed.
Can  you help please.
Regards Barry

Harvie, 2021-01-09:
Hello. Have you installed bCNC using pip as reccomended? Can you post screenshot of your problem?

GitHubCaps, 2021-01-09:
@Barry123571 You may need to change the Timeout (top right corner) from default (5) to a larger value. I would start with (inf) which is infinite.