vlachoudis/bCNC #1511: Heightmap plugin exception raised

creat1c, 2021-01-09:
I am trying to work with a heightmap image but seemingly the plugin aborts with the message shown below. I've attached the image and screenshots of the settings as well.

`Traceback (most recent call last):
  File "c:\python37\lib\site-packages\bCNC\Utils.py", line 465, in __call__
    return self.func(*args)
  File "c:\python37\lib\site-packages\bCNC\lib\tkExtra.py", line 866, in handleK
ey
    active = unicode(self.get(ACTIVE))
NameError: name 'unicode' is not defined
Traceback (most recent call last):
  File "c:\python37\lib\site-packages\bCNC\Utils.py", line 465, in __call__
    return self.func(*args)
  File "c:\python37\lib\site-packages\bCNC\lib\tkExtra.py", line 866, in handleK
ey
    active = unicode(self.get(ACTIVE))
NameError: name 'unicode' is not defined
Traceback (most recent call last):
  File "c:\python37\lib\site-packages\bCNC\Utils.py", line 465, in __call__
    return self.func(*args)
  File "c:\python37\lib\site-packages\bCNC\lib\tkExtra.py", line 866, in handleK
ey
    active = unicode(self.get(ACTIVE))
NameError: name 'unicode' is not defined
Traceback (most recent call last):
  File "c:\python37\lib\site-packages\bCNC\Utils.py", line 465, in __call__
    return self.func(*args)
  File "c:\python37\lib\site-packages\bCNC\lib\tkExtra.py", line 866, in handleK
ey
    active = unicode(self.get(ACTIVE))
NameError: name 'unicode' is not defined
Traceback (most recent call last):
  File "c:\python37\lib\site-packages\bCNC\Utils.py", line 465, in __call__
    return self.func(*args)
  File "c:\python37\lib\site-packages\bCNC\lib\tkExtra.py", line 866, in handleK
ey
    active = unicode(self.get(ACTIVE))
NameError: name 'unicode' is not defined
Traceback (most recent call last):
  File "c:\python37\lib\site-packages\bCNC\Utils.py", line 465, in __call__
    return self.func(*args)
  File "c:\python37\lib\site-packages\bCNC\ToolsPage.py", line 1746, in execute
    self.tools.getActive().execute(self.app)
  File "c:\python37\lib\site-packages\bCNC\plugins\heightmap.py", line 326, in e
xecute
    MAT.pad_w_zeros(TOOL)
  File "c:\python37\lib\site-packages\bCNC\lib\imageToGcode.py", line 823, in pa
d_w_zeros
    temp[to:to+w, to:to+h] = self.matrix
TypeError: slice indices must be integers or None or have an __index__ method`

[image]
[image]
[image]

GitHubCaps, 2021-01-09:
@creat1c Are you using the most recent version of bCNC?

I just ran this without a problem using Python 3.8.2

[image]

output using your endmill settings:
```gcode
(Block-name: Header)
(Block-expand: 0)
(Block-enable: 1)
M3 S12000
G4 P3
G0 Z10
(Block-name: test)
(Block-expand: 1)
(Block-enable: 1)
(Size: 100 x 45 x 1)
(Endmill shape: Ball End , Diameter: 2.400)

G0 Z3.0000
F1200.0000
F500.0000
G0 X27.6872 Y44.9127
F1200.0000
G1 Z-0.1289
 X28.0479 Z-0.4084
 X72.1489
 X72.5097 Z-0.1289
F500.0000
G0 Z3.0000
 X72.7803 Y44.0109
F1200.0000
G1 Z-0.1027
 X72.2391 Z-0.5120
 X66.7378 Z-0.5154
 X27.9577 Z-0.5120
 X27.4166 Z-0.1027
F500.0000
G0 Z3.0000
 X27.1460 Y43.1090
F1200.0000
G1 Z-0.1027
 X27.5068 Z-0.2896
 X72.6901
 X73.0508 Z-0.1138
F500.0000
G0 Z3.0000
 X73.7723 Y42.2071
F1200.0000
G1 Z-0.1138
 X73.3214 Z-0.2825
 X26.8755
 X26.4246 Z-0.1138
F500.0000
G0 Z3.0000
 X25.3423 Y41.3053
F1200.0000
G1 Z-0.1027
 X25.8834 Z-0.2825
 X74.3134 Z-0.2789
 X74.8545 Z-0.1138
F500.0000
G0 Z3.0000
 X85.9474 Y40.4034
F1200.0000
G1 Z-0.1289
 X85.6769 Z-0.2825
 X14.5200
 X14.2494 Z-0.1289
[... 533 lines cut ...]
G0 Z3.0000
 X72.5097 Y0.7215
F1200.0000
G1 Z-0.1289
 X71.9686 Z-0.6115
 X71.3373 Z-0.6128
 X28.2283 Z-0.6120
 X27.6872 Z-0.1289
F500.0000
G0 Z3.0000
 X27.8675 Y0.0000
F1200.0000
G1 Z-0.1289
 X28.3185 Z-0.5248
 X29.4909 Z-0.5422
 X71.6980 Z-0.5388
 X72.2391 Z-0.1289
G0 Z3.0000

(Block-name: Footer)
(Block-expand: 0)
(Block-enable: 1)
G0 Z10
M5
```

GitHubCaps, 2021-01-09:
Test with conical bit gives me this
[image]

creat1c, 2021-01-10:
I didn't use 3.8.2 initially, but now verified the behaviour is similar (see below). Also the version of bCNC is 0.9.14 which is what was installed by `py -m pip install bCNC`

From the error is seems like there might be an issue with the end mill settings? I tried using the exact settings in the screenshot but to no avail.

Also, i tried this on my windows machine as well as the rPI connected to my mill, with comparable results.

`
Traceback (most recent call last):
  File "E:\Python\3.8.2\lib\site-packages\bCNC\Utils.py", line 526, in __call__
    return self.func(*args)
  File "E:\Python\3.8.2\lib\site-packages\bCNC\ToolsPage.py", line 1747, in exec
ute
    self.tools.getActive().execute(self.app)
  File "E:\Python\3.8.2\lib\site-packages\bCNC\plugins\heightmap.py", line 326,
in execute
    MAT.pad_w_zeros(TOOL)
  File "E:\Python\3.8.2\lib\site-packages\bCNC\lib\imageToGcode.py", line 823, i
n pad_w_zeros
    temp[to:to+w, to:to+h] = self.matrix
TypeError: slice indices must be integers or None or have an __index__ method
`

creat1c, 2021-01-10:
I was able to get the plugin to work by changing imagetogcode.py, line 823 to read

`self.t_offset = int((ts-1)/2)`

GitHubCaps, 2021-01-10:
Honestly, this is very puzzling. I haven't seen any issues with the heightmap plugin or its dependency image2gcode before and like I said, it works for me without modifying. I do know that there are some improvements to be made for both, but nothing that has stopped it from running before. In my opinion, there seems to be varying results by users, but why? I tested in py2 and py3 and it simply worked.

creat1c, 2021-01-10:
The issue seemed to be that this line was previously (without the cast to an int) was evaluated to a float. This seemingly caused and issue with the indexing. All I did was to add the cast to an int and now it works.

As to why this happened, i am unsure. Within my debugger, ts had the value 3 so (3-1)/2=2 which should be an int?

If you have any ideas how I can test further, ill gladly to it, to ensure this does not happen for anyone else.

GitHubCaps, 2021-01-10:
@creat1c My apologies! Just realized that I used my own image2gode instead of imageToGcode.

I have a fixed version that doesn't cast to integer, but instead uses `self.t_offset = (ts-1)//2`
I will put it here for you to try. Hopefully soon, I will commit/push it here.

This one fixes all deprecation issues for py2 and py3!
[imageToGcode.py.zip](https://github.com/vlachoudis/bCNC/files/5793118/imageToGcode.py.zip)

creat1c, 2021-01-10:
Ah! Thanks for checking.
One more question I had - is the step size normally dependant on the milling bit? It used a very fine grid (0.1 mm) instead of the milling bit i selected (1mm) and I was wondering if my fix has some side effect?

GitHubCaps, 2021-01-10:
The step size is in percent of the amount to move the bit over. For example, 40.0 moves the bit by 40% on next pass. I have a better heightmap plugin that allows for using both finish and roughing bits and does not really on the endmill database. It also has finer grain control. Give me a little while and I will put it and the actual lib I use here for you to test out,  it is way better (imho) than what is provided in bCNC (it just has a basic implementation)

GitHubCaps, 2021-01-10:
The percent is in relation to bit size (diameter)

GitHubCaps, 2021-01-10:
@creat1c BTW, could you reopen this until it's committed? This still effects py3 ATM

creat1c, 2021-01-10:
Sure.

If you need someone to betatest the new script, I can do that.