vlachoudis/bCNC #1512: Installation instruction

InnoGreenTech, 2021-01-09:
Hi, i hope your are doing well,

To install bCNC you propose to use "pip to install". I thinck that would be well to change it by "pip3 install" else bCNC works on python 2.X

Best regards.

GitHubCaps, 2021-01-10:
That really depends on the end user knowing their setup.

Most recent OSes now come with Python3 as default. The manual did have instructions for pip/pip2 before, however, it really gets confusing to end users, especially when moving forward to python3 with the code. Not to mention that there isn't a pip3 on newer OSes that don't support python2 anymore. 

Probably best to add a reference to pip for beginners, which explains how to check which pip/python, instead of making the manual any more complicated.

InnoGreenTech, 2021-01-12:
Yes , that's right, that's depend of the user.

Thanks

GitHubCaps, 2021-01-14:
@InnoGreenTech Please close this.

I have added:
# Before you begin, read this!
https://packaging.python.org/tutorials/installing-packages/

We are not responsible if you mess up your system!
You have been WARNED!

to https://github.com/vlachoudis/bCNC/wiki/Installation
and also to README.md, but that will be in a future commit.

Thanks