vlachoudis/bCNC #1515: GRBL End-Of-Life release?

kfmut, 2021-01-13:
Hi,

I’ve noticed that bCNC is reorienting towards newer grblHAL https://github.com/vlachoudis/bCNC/issues/1428 so maybe it’s a good time to make some kind of GRBL End-Of-Life release?

I've a working setup with GRBL 1.1f guided by bit dated version of bCNC and have no plans to upgrade it to grblHAL in near or in not so near future. So my concern is that if something bad happens with my current setup it would be relatively difficult to recover from that if bCNC somehow will lose backward compatibility with plain GRBL.

Harvie, 2021-01-13:
Don't worry. There are currently no plans to end support of GRBL. I still use GRBL, because my grblHAL setup is not ready yet. And we will surely support it. There are plans to support as much controllers as possible.

But i still take the freedom to strongly reccomend the grblHAL to new users, because it's worth it in the long run. Just go and try :-)

closed 2021-01-13 by Harvie

kfmut, 2021-01-14:
@Harvie , thank you for quick response!