vlachoudis/bCNC #1517: Alarm 4, tool change

fkfr, 2021-01-14:
Hello, my first tool change test. I don't know how to solve it. Can you help me?

[image]
[image]

Harvie, 2021-01-14:
`“ALARM:4” : _(“Probe fail. The probe is not in the expected initial state before starting probe cycle, where G38.2 and G38.3 is not triggered and G38.4 and G38.5 is triggered.”),`

Are you trying to do toolchange with probing? i've never tried this...

fkfr, 2021-01-15:
I am confused about the procedure to follow for the tool change.
1) do the 0 WCS => probe tab => probe.
2) select the 'position' of the tool change. => probe tab => tool => change => get.
3) select the 'position' of the probe. => probe tab => tool => probe => get.
and then?

tk.