vlachoudis/bCNC #1520: Debian 10 no jogging

NickPavel, 2021-01-18:
This is on Debian 10.
It connects to the grbl 0.9j CNC, but it would not do any jogging.

`nick@cnc:~$ sudo pip install bCNC
Collecting bCNC
Requirement already satisfied: opencv-python==4.2.0.32; "arm" not in platform_machine and "aarch64" not in platform_machine in /usr/local/lib/python2.7/dist-packages (from bCNC) (4.2.0.32)
Requirement already satisfied: Pillow>=4.0 in /usr/local/lib/python2.7/dist-packages (from bCNC) (6.2.2)
Requirement already satisfied: pyserial; sys_platform != "win32" in /usr/local/lib/python2.7/dist-packages (from bCNC) (3.5)
Requirement already satisfied: numpy>=1.12 in /usr/lib/python2.7/dist-packages (from bCNC) (1.16.2)
Installing collected packages: bCNC
Successfully installed bCNC-0.9.14.312
nick@cnc:~$ python -m bCNC
new-config Utils <ConfigParser.ConfigParser instance at 0x7f6bc1e1ae60>
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
self.canvas.winfo_id(): 35651674
Not Loading 6 axis displays
Not Loading 6 axis displays
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/Sender.py", line 787, in serialIO
    elif self.mcontrol.parseLine(line, cline, sline):
  File "/usr/local/lib/python2.7/dist-packages/bCNC/controllers/_GenericController.py", line 213, in parseLine
    self.parseBracketAngle(line, cline)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/controllers/GRBL0.py", line 27, in parseBracketAngle
    CNC.vars["wx"] = float(pat.group(5))
TypeError: float() argument must be a string or a number

Opening in existing browser session.
`

boris-so, 2021-02-22:
> It connects to the grbl 0.9j CNC, but it would not do any jogging.

Jogging mode has been added in 1.1 version (see [this](https://github.com/gnea/grbl/wiki#new-features-in-v11)). I bielieve u should flash a newer grbl version to your controller.

Or try to set `Controller` option in `File` tab to value `GRBL0`. But that version is obsolete and re-flashing recommended.

PS. I am on debian 10 too and bCNC from pip (0.9.14-dev) and GRBL 1.1 - works great :)

Harvie, 2021-02-22:
@NickPavel Can you test upgrading to latest version using pip? Grbl 0.9 support was fixed few days ago.

> Jogging mode has been added in 1.1 version

This is obviously not true. bCNC should be able to jog GRBL 0.9. How do you think people would use their machines without joging? :-D grbl 1.1 added more responsive realtime jogging controls. but bCNC does not rely on this feature.

boris-so, 2021-02-23:
Sorry for unclear explaination(
My point was to check bCNC `Controller` parameter since grbl < 1.1 [use G-codes for jogging](https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#jogging) while grbl >= 1.1 added special $J command for that.
So bCNC may send newer commands to older controller.

Or re-flash controller to grbl 1.1

Harvie, 2021-02-23:
> So bCNC may send newer commands to older controller.

When i was implementing this feature i've been careful enough to only add it to GRBL1 plugin while leaving functionality of GRBL0 plugin intact.