vlachoudis/bCNC #1521: BCNC moves to mpos if probe is stopped in middle before touching the base

manucool, 2021-01-26:
Hi,

I am facing following issue while using bCNC.

1) If the robing is stopped in middle then CNC starts to return to zero x and y axis as set in MPOS not in WPOS also it does not make the z axis up while doing so. This leaded to breakage of bit. 

using GRBL HAL 1.1

please tell whether this can be changed or not.

Also, is there any option by which i can make the MPOS zero at the starting of machine.

Regards
Manindera Singh
SSJB Automation

c-, 2021-01-26:
I had this happen a couple of times lately. Seems that it issues a G53 G0 X0 Y0 Z0 after the probe is aborted, and bad things happen (the latest one snapped a chunk off my vacuum pickup). At least I don't break bits, because I use an already-broken drill for probing.

manucool, 2021-01-26:
Thanks for replying.

I wanted to know whether this can be resolved by some means. 

Also there are three more queries

1) if there is any option to save the probe data of auto level mode if the auto level probe is stopped in middle.

2) Can we make mpos zero at any time or a button is needed for doing so. 
 
3) there is one problem which I faces when I zoom in the probe data while making auto level probing the processing of the application gets increased and my PC hangs. I have to use task manager to exit the program. Normally It takes 100mb of ram approx but during that problem it start using 2.5 gb if ram 

Please help to sort out these queries

Rest the software bcnc is very much useful which I liked after using UGS and various other applications.

c-, 2021-01-26:
I'm not actually one of the devs; I just happened to stop in to see if there was an answer to this problem and this was at the top of the issue list.

c-, 2021-01-26:
looks like a duplicate of #1389