vlachoudis/bCNC #1524: Z Moves wrong at start

NFTVirus, 2021-01-30:
Hello,

since a few days I have issues with my mill. When I start a code the first thing bCNC does is to moves the Z axis several cm down into the material. Afterwards it moves it with this z value to the position where it should start milling. After this it looks to work correct. This issue occured also with old g Code which always worked great. I also checked that all axis were zeroed correct but it also visualizes the movement in material. Has anyone an idea what's going wrong?

Bestr regards

Christian

kfmut, 2021-02-12:
@NFTVirus have you solved that problem? If not then attach g-code file it will be huge help.

NFTVirus, 2021-02-13:
@kfmut I could not solve the issue up to now. I check later far the good, but I am quite sure that the code does not cause the issue. I tried it now with bCNC on a windows laptop and it worked perfect. Also I. The preview of the maschine paths everting ja fine but when it travels to the position. Where it should touch the first time the stock it travels in much too deep ways several cm inside the storm

karoria, 2021-02-13:
@NFTVirus This behaviour is due to no proper setting of G28 safe retract. You can simply remove G28 line from program startup and run the job. If you are posting the program through fusion 360, (following instruction is as per mac, windows app can be little different) at the post process dialog box, expand properties tab, there you will find safe retracts value which should be G28 now which is causing problems, instead change it with "clearance height" and your program is good to go.

kfmut, 2021-02-13:
If it really relates to G28 it's better to set G28 position with G28.1, huge help in avoiding stupid mistakes and keeping some endmils safe.

NFTVirus, 2021-02-13:
first of all thanks for the help! I will check it now on the mill. Start of my code looks like follows:

`G90 G94
G17
G21

G28 G91 Z0
G90

(2D-Kontur2)
T4
S12000 M3
G54
M8
G0 X106.18 Y116.939
Z15
G1 Z5 F500
Z4.4 F50
Z-3
X106.187 Z-3.078
X106.21 Y116.94 Z-3.153
X106.247 Y116.941 Z-3.222
X106.297 Y116.942 Z-3.283
X106.357 Y116.943 Z-3.333
X106.427 Y116.945 Z-3.37
X106.502 Y116.947 Z-3.392
X106.58 Y116.949 Z-3.4
X106.979 Y116.959`

NFTVirus, 2021-02-13:
I uploaded a video of the issue. You see the xz view of in bCNC and how the maschine runs starting from the zero postion

[https://youtu.be/c5dxvkZfhXQ]

NFTVirus, 2021-02-13:
So I can confirm that deleting the g28 code solved the issue. But honestly I do not really understand what's going wrong there. I also did not find the setting in Fusion but that's something I will figure out. Again thanks a lot for the help!

kfmut, 2021-02-13:
`G28 G91 Z0` move is relative to special G28 position which permanently stored in GRBL:

https://drive.google.com/file/d/1cjXnK89BSdK0vOiSXt5O_Zrdy0mIbAFl/view?usp=sharing

By default G28 has machine coordinates of X0 Y0 Z0, so by issuing `G28 G91 Z0` GRBL moves to machine zero.

If you have working homing and repeated machine zero you could move your spindle to some safe point and issue `G28.1` command to store that safe position within GRBL. And that first `G28 G91 Z0` move will ALWAYS lift router head before moving it to the first point of toolpath.

If homing isn't an option just switch off in CAM or remove `G28 G91 Z0` move from code.

NFTVirus, 2021-02-13:
Thank you for the explanation now I got it!

closed 2021-02-13 by NFTVirus