vlachoudis/bCNC #1526: Program crahes when turning on webcam (py3 @ win7)

ldpgh, 2021-01-31:
Dry test
... WebCAM already connected
... just start bCNC with no GRBL controller connected
... CLick on the icon  "Toggle Display of Camera"
The program tries to display the stream, but after few seconds the crash form comes up.

... help(cv2.version)   returns    4.4.0.46
... python = Python 3.7.4 (tags/v3.7.4:e09359112e, Jul  8 2019, 19:29:22) [MSC v.1916 32 bit (Intel)] on win32

ldpgh, 2021-01-31:
>>python -m bCNC
new-config Utils <configparser.ConfigParser object at 0x02D287F0>
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
self.canvas.winfo_id(): 2820732
Using old style comports()!
Using old style comports()!
Using old style comports()!
Not Loading 6 axis displays
Not Loading 6 axis displays
[ WARN:0] global C:\Users\appveyor\AppData\Local\Temp\1\pip-req-build-tts2sm8m\opencv\modules\videoio\src\cap_msmf.cpp (677) CvCapture_MSMF::initStream Failed to set mediaType (stream 0, (640x480 @ 30) MFVideoFormat_RGB24(codec not found)
[ WARN:0] global C:\Users\appveyor\AppData\Local\Temp\1\pip-req-build-tts2sm8m\opencv\modules\videoio\src\cap_msmf.cpp (677) CvCapture_MSMF::initStream Failed to set mediaType (stream 0, (640x480 @ 30) MFVideoFormat_RGB24(codec not found)
[ WARN:0] global C:\Users\appveyor\AppData\Local\Temp\1\pip-req-build-tts2sm8m\opencv\modules\videoio\src\cap_msmf.cpp (677) CvCapture_MSMF::initStream Failed to set mediaType (stream 0, (176x144 @ 30) MFVideoFormat_RGB24(codec not found)

>>
>>

ldpgh, 2021-01-31:
USB Device Tree Viewer output ....

    =========================== USB Port2 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 2-2

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB-Verbundgerät
Device Path              : \\?\usb#vid_058f&pid_5608#5&19fc2e83&0&2#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Device ID                : USB\VID_058F&PID_5608\5&19FC2E83&0&2
Hardware IDs             : USB\VID_058F&PID_5608&REV_0004 USB\VID_058F&PID_5608
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0160 (GUID_DEVCLASS_USB)
Driver                   : system32\DRIVERS\usbccgp.sys (Version: 6.1.7601.22535  Date: 2015-03-07)
Driver Inf               : C:\Windows\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Interface GUID           : {a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0002.Hub_#0001
Location IDs             : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(2)
Container ID             : {669d8898-d6d9-11e9-bee9-34e6d7456159}
Manufacturer Info        : (Standard-USB-Hostcontroller)
Capabilities             : 0x84 (Removable, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 2
Power State              : D0 (supported: D0, D3, wake from D0)
 Child Device 1          : USB 2.0 WebCamera (USB-Videogerät)
  DevicePath             : \\?\usb#vid_058f&pid_5608&mi_00#6&2df3a673&0&0000#{6994ad05-93ef-11d0-a3cc-00a0c9223196}\global
  KernelName             : \Device\000001da\global
  Device ID              : USB\VID_058F&PID_5608&MI_00\6&2DF3A673&0&0000
  Class                  : Image

ldpgh, 2021-01-31:
USB Device Tree Viewer output .... 2nd part

        ---------------- Connection Information ---------------
Connection Index         : 0x02 (2)
Connection Status        : 0x01 (DeviceConnected)
Current Config Value     : 0x01
Device Address           : 0x07 (7)
Is Hub                   : 0x00 (no)
Number Of Open Pipes     : 0x02 (2)
Device Bus Speed         : 0x02 (High-Speed)
Pipe0ScheduleOffset      : 0x00 (0)
Pipe1ScheduleOffset      : 0x00 (0)
Data (HexDump)           : 02 00 00 00 12 01 00 02 EF 02 01 40 8F 05 08 56   ...........@...V
                           04 00 03 01 00 01 01 02 00 07 00 02 00 00 00 01   ................
                           00 00 00 07 05 82 03 20 00 07 00 00 00 00 07 05   ....... ........
                           81 05 00 14 01 00 00 00 00                        .........

    ---------------------- Device Descriptor ----------------------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x01 (Device Descriptor)
bcdUSB                   : 0x200 (USB Version 2.00)
bDeviceClass             : 0xEF (Miscellaneous)
bDeviceSubClass          : 0x02
bDeviceProtocol          : 0x01 (IAD - Interface Association Descriptor)
bMaxPacketSize0          : 0x40 (64 bytes)
idVendor                 : 0x058F (Alcor Micro, Corp.)
idProduct                : 0x5608
bcdDevice                : 0x0004
iManufacturer            : 0x03 (String Descriptor 3)
 Language 0x0409         : "Alcor Micro, Corp."
iProduct                 : 0x01 (String Descriptor 1)
 Language 0x0409         : "USB 2.0 WebCamera"
iSerialNumber            : 0x00 (No String Descriptor)
bNumConfigurations       : 0x01 (1 Configuration)
Data (HexDump)           : 12 01 00 02 EF 02 01 40 8F 05 08 56 04 00 03 01   .......@...V....
                           00 01                                             ..

ldpgh, 2021-01-31:
No issue when I try to open the WebCam with VLC (Version 3.0.8). Works as expected.

ldpgh, 2021-02-06:
program version ....  bCNC  0.0.14-dev  (win32 py3.7.4)

ldpgh, 2021-02-06:
[image]

Harvie, 2021-02-16:
This looks pretty much like OpenCV bug, as such it belongs in here: https://github.com/opencv/opencv/issues

closed 2021-02-16 by Harvie