vlachoudis/bCNC #1527: Calibration value cannot be manually entered

ExploratoryDevices, 2021-02-01:
I have a permanently mounted tool height setter. I want to be able to reliably set my Z zero to be my bed without having to touch off on the bed. I have saved the calibration value for this (effectively the trigger height of my tool setter) and tried to manually type it into the calibration field, but it doesn't appear to update when I change it this way. Running a tool change gives the same result for TLO regardless of what number I've entered into the calibration field. It only changes if I actually run the calibration macro.

It seems like it may need a "set" button like the TLO box does. Or perhaps I'm doing it wrong?

Ogurezzz, 2021-02-05:
Did you tried to press "Zero" button in autolevel section?
It will set current position as 0.000 and recalculate all other points of calibration net.

ExploratoryDevices, 2021-02-06:
Thanks for the reply. I've never used the auto-level section. I'm referring to the "tool" section and don't want to run a calibration cycle at all. I want to use a calibration value I recorded from a previous calibration session so that all TLO's are calculated based on a common value for certain operations (without being subject to error in subsequent setting of the zero position and recalibration).