vlachoudis/bCNC #1529: Z Homing Error

willc2003, 2021-02-06:
Z, and only Z axis, isn’t homing correctly.  It adds an extra “skip” at the end of the homing cycle. So if I set Z=0, then travel 20mm, it will travel back 25mm.   Also interesting, my $102 is about double my $100/$101.  All the equipment is the same so I would think the calibration would have been the same?  The $100-$102 allow me to accurately travel a measured distance.  

 [](https://youtu.be/-kUrQQVB1qo) 
[image]

willc2003, 2021-02-06:
https://youtu.be/-kUrQQVB1qo

Danivas13, 2021-02-08:
I think that the problem is: the z axis motor is disabled when the movement end and the weight pull it down.

try: $1=255