vlachoudis/bCNC #1532: Compensate for axes not aligned at 90 °

Danivas13, 2021-02-09:
Hi, what do you think about implementing some function like this? 
we could designate one axis as master and adjust the other just before sending it to arduino.

just before sending any gcode. and keeping the illusion (in the interface) that the machine is aligned.

This example is to compensate X axis misalignment.

[image]

Harvie, 2021-02-09:
If you really can't fix your machine geometry, you should handle this in motion controller.

Danivas13, 2021-02-10:
sorry for my english i don't understand what you mean by motion controller ,gbrl?