vlachoudis/bCNC #1536: Bug in Path.split2contours function ?

DodoLaSaumure, 2021-02-11:
As discussed in #1535 with @Harvie I am afraid I have discovered a bug in
```python
Path.split2contours(self, acc=EPSV)
```
even if I am not sure yet.

When I make a pocket with inside island on file triangle4.dxf herewith (zipped to allow import inside github comment), I get weird result when I activate the option 
`allow G1 linking segments(default yes)`

When disabled, weird segments disappear.
I use a diam 2 or diam 5 endmill to reproduce.
Here's the file where some problems occur
[triangle4.zip](https://github.com/vlachoudis/bCNC/files/5966840/triangle4.zip)

- With option enabled I get this result with a diam 5 endmill
[image]

- With option disabled I get this
[image]

I think that the bug occurs when calling split2contous in file pocketwithislands.py, lines 75 -81
```python
			#concatenate newpath in a single list and split2contours
			if allowG1 :
				MyFullPath = Path("Pocket")
				for path in newpathList :
					for seg in path:
						MyFullPath.append(seg)
				newpathList = MyFullPath.split2contours()
```

Here are other files where some problems occur with pockets, **both with the old pocket function** and the **new pocketwithislands plugin**.
[horloge.zip](https://github.com/vlachoudis/bCNC/files/5966995/horloge.zip)

- Making a pocket with the standard pocket tool on the C of Besançon provides this, with an diam 1mm endmill
[image]

and there are other problems on some letters, with both tools.

In addition, if this can help, 
```bash
grep -r split2contours 
```
gives the following results

```python
lib/bpath.py:	def split2contours(self, acc=EPSV):
lib/bpath.py:			opath = opath.split2contours()
lib/bpath.py:	# FIXME: This probably can be replaced with split2contours() and i've just reinvented wheel LOL

plugins/difference.py:				#paths_newbase.extend(newbase.split2contours())

plugins/pocketwithislands.py:			#concatenate newpath in a single list and split2contours
plugins/pocketwithislands.py:				newpathList = MyFullPath.split2contours()
plugins/pocketwithislands.py:		purgednewoutpath= newoutpath.split2contours()#list of paths
plugins/pocketwithislands.py:		cleanpath = cleanpath.split2contours()

CNC.py:			opath = path.split2contours(0.0001) #Lowered accuracy due to problems interfacing arcs and lines in DXF
CNC.py:		opath = opath.split2contours()
CNC.py:				opath = opath.split2contours()
CNC.py:				opath = opath.split2contours()
```

DodoLaSaumure, 2021-02-11:
No it is not in split 2 contours that the problem occurs
with commenting ,newpathList = MyFullPath.split2contours()
```python

			if allowG1 :
				MyFullPath = Path("Pocket")
				for path in newpathList :
					for seg in path:
						MyFullPath.append(seg)
# 				newpathList = MyFullPath.split2contours()
		if newpathList:
```

 I ve got the following result

[image]
and the problem is in adding segment lines in 
```python
class PocketIsland:
```
but I do not see why yet.
Pierre

Harvie, 2021-02-11:
I have issue even with G1 linking disabled. It does milling inside of islands quite consistently:
[image]

Original shape with islands:

[image]

MARIOBASZ, 2021-02-12:
Really impressed. I will try understand how it works. 
With bit 0.7 this problem occurs. 
[image]

MARIOBASZ, 2021-02-12:
Maybe the option “Consider all other shapes as islands “ is useful?

DodoLaSaumure, 2021-02-12:
> I have issue even with G1 linking disabled. It does milling inside of islands quite consistently:

@Harvie 
Can you please send your file ?
Possibly in this case it is another issue related to the following lines

```python
			path.directionSet(1) #turn path to CW (conventional when milling inside)

			D = path.direction()
			if D==0: D=1
```
```python
			dir = path.direction()
			opath = path.offset(offset*float(dir))
```
```python
			dir = island.direction()
			offIsl = island.offset(-offset*float(dir))
```

@MARIOBASZ 
you have to toggle islands previously to apply pocket islands; I do not see the island toggled in the picture you sent.
I have limited recurse depth to 100, maybe this can explain why the 0.7 bit does not go deeper.

DodoLaSaumure, 2021-02-12:
update

@Harvie 
I think that I have reproduced your problem and fixed it by adding the following lines in the code on line 162 and following
```python
			if self.depth >0 and p3 is not None and p4 is not None :
				self.islandG1SegList.append(Segment(Segment.LINE,p3,p4))
			offIsl.intersectSelf()
			offIsl.removeExcluded(island, abs(offset))
```

with the patch
[image]

without the patch
[image]

All the best
Pierre

but the intial problem remains
[image]

DodoLaSaumure, 2021-02-12:
I have added a fix for @Harvie issue, if you kindly can give a feedback.

> I have issue even with G1 linking disabled. It does milling inside of islands quite consistently:

Thanks
Pierre

Harvie, 2021-02-12:
> I think that I have reproduced your problem and fixed it 

Cool. thanks. i will try it. Also i've noticed on the image you've sent that there is unnecessary cutting of the air in there:

[image]

At least it seems to me that some areas are cut twice. Don't get me wrong this is still pretty awesome. But fixing it would certainly make sense.

Harvie, 2021-02-12:
Here is the file i've used:
[pocket_island_bug.ngc.txt](https://github.com/vlachoudis/bCNC/files/5971689/pocket_island_bug.ngc.txt)

DodoLaSaumure, 2021-02-12:
Hi Tomas

I've tried out your file bug with Diam 1mm endmill
with option allowG1 disabled I've got the following
[image]
with option allowG1 enabled
[image]

So your bug seems to be solved by PR #1537
But the original bug remains because G1 segs linking pocket paths are sometimes interesect island when jumping from an area to another (Most probably)
I will try to add a fix to this issue this week end.

DodoLaSaumure, 2021-02-12:
> Also i've noticed on the image you've sent that there is perhaps unnecessary cutting of the air in there. Or maybe i am wrong and its just rapids fused together on screen.

I think that's an artefact in the image due to resolution of the picture posted.
But the file seems ok
Here's a zoom in
[image]

Harvie, 2021-02-12:
[image]

DodoLaSaumure, 2021-02-12:
aha 
intersectSelf
removeExcluded
&
split2contours 

are the real MVP ;)

DodoLaSaumure, 2021-02-13:
Please find my test cases for py2 & py3
[TestCases.zip](https://github.com/vlachoudis/bCNC/files/5975231/TestCases.zip)

I think it's stable now.
Pierre

DodoLaSaumure, 2021-02-13:
[extraTestCases.zip](https://github.com/vlachoudis/bCNC/files/5975245/extraTestCases.zip)

Now I have not anymore test case failing, I think  this issue can be considered as solved by the last fix in  #1538

DodoLaSaumure, 2021-02-13:
Update : Just to share my view and ask community : 

I think pocketwitisland experimental plugin is now more stable than standard former pocket tool, and more powerful.
The reason why I say this is because i have found bugs in pocket standard tool, that not occur in the new one.

For example test with the C of "Besançon" in extraTestCases
It fails on pocket standard tool
and passes on pocketwithisland experimental plugin.

and there are many other cases.

Should we consider replacing the standard tool with the experimental one ?

Another thing is that I have seen (but not sure) :  there might remain bugs in the "profile" tool, making an offset on some shapes sometimes fails (to be confirmed)

Harvie, 2021-02-13:
> Should we consider replacing the standard tool with the experimental one ?

Yes, i've suggested this. However i would like the new plugin to be able to replace both pocket and profile... Can you add some "depth" limit to your plugin? So we can specify how many offset passes should be done...

Eg.:
0 = infinity = do full pocket
1 = single offset = just as classic bCNC profile
3 = do three offsets

closed 2021-02-13 by Harvie

Harvie, 2021-02-13:
I've just tested and i have some issues with uneven spacing between passes:

[image]

DodoLaSaumure, 2021-02-13:
Hi @Harvie thank you for your remarks

> i would like the new plugin to be able to replace both pocket and profile... Can you add some "depth" limit to your plugin? So we can specify how many offset passes should be done...

> Eg.:
0 = infinity = do full pocket
1 = single offset = just as classic bCNC profile
3 = do three offsets

Very good idea !!
However, this raises some (small) questions.

- 1 = single offset = just as classic bCNC profile

yes, but we must add a direction  parameter (inside / Outside profile ) in this only case. Nothing difficult. We just have to think ergonomy for not bothering user with this parameter in the other cases. We also have to fix a bug that sometimes seems to occur in offset function.

- 0 = infinity = do full pocket
what do you mean by infinity ? on my old laptop
```python
print (sys.getrecursionlimit())
1000
```
I think 1000 is much too big.
Should we consider that 100 is a reasonable max limit ? What are your opinions on this ?

- 3 = do three offsets
Tell me if I m wrong but I do not see any use case for "3" or "5" offsets. However, if  we limit infinity to 100 passes, this could be a way going deeper, taking as parameter max ( sys.getrecursionlimit(), n)

- Uneven spacing
> I've just tested and i have some issues with uneven spacing between passes

I was aware of this. I think this not a real issue for most users, this can be easily fixed by changing Stepover % parameter in the mill parameters. Maybe we can make a fix later on this particular point, but this is not so important, I think.

1mm bit, G1 enabled 10% Stepover % parameter
[image]

DodoLaSaumure, 2021-02-13:
Update

> this can be easily fixed by changing Stepover % parameter in the mill parameters

Setting a stepover lower than 50% guarantees that all the material will be removed.

Some small areas can remain uncut if stepover is > 50%

Harvie, 2021-02-13:
> yes, but we must add a direction parameter (inside / Outside profile ) in this only case.

profile operation currently adds "in/out" tag in block name, so cut operation later can cut them in proper direction. but i don't think it can currently handle situation with directions being mixed in single block. So probably there will have to be "conventional/climb" setting for the pockets with islands... i don't know... i remember this was very annoying when it didn't worked out properly out of box. and i don't really want to break it again.

> Tell me if I m wrong but I do not see any use case for "3" or "5" offsets. 

More than one offset is very usefull to provide more space to evacuate the chips to prevent recutting, protect the milling bit and be able to achieve better surface finish. I've broken way too many bits because lack of this feature :-)

DodoLaSaumure, 2021-02-13:
> profile operation currently adds "in/out" tag in block name, so cut operation later can cut them in proper direction. but i don't think it can currently handle situation with directions being mixed [...]

- I do not see major issue about this.
- it's easy to get direction from a path
```python
dir = path.direction()
```
- it's easy to set a new one
```python
Path.directionSet(self, opdir)
```

- In addition to this, as far as I know, but maybe I am wrong, I learned that the direction of machining does not matter very much on a CNC, compared to a manual ( CNC have ball screws compensating the loose, and there is no  risk of being aspired by the mill)

- more problematic is the overcut parameter, but this is not really an issue except that too many parameters can confuse users.

> I've broken way too many bits because lack of this feature :-)
Really ?
To limit risks of breaking small bits, I calibrate my surface in Z with the Z mapping tool and make an approach z pass, starting over the surface at z>0
Anyway, the first time bit enters the material with recurse depth 0, will be a plain bit milling.
But I agree with you it's not a problem allowing a recurse depth of n.

In conclusion, before making any correction, **I would like we agree on the scope previously.**

What I suggest : 
- Profile function : Allow recurse depth 0 for profile, and get the direction of the path, set it at user's convenience ( a flip direction toogle ?), allow overcut parameter for recursive depth Zero only, allow inside/outside parameter.
- Allow recurse depth manually set to any n
- Allow full recurse, limited to a max (I suggest 100)
-  add segments where uneven lines happen in the pockets with islands, to be sure not to let small remaining areas in the pocket, even with stepover > 50%
- make it ergonomic and easy to understand for users.

Do you agree with this scope ?
Thanks 
Pierre

Harvie, 2021-02-13:
> To limit risks of breaking small bits, I calibrate my surface in Z with the Z mapping tool and make an approach z pass, starting over the surface at z>0

It is not all about Z depth step at all... You will break bits anyway in some cases. In some setups you just can't cut deep grove in the same width as the milling bit. If you are 3 centimeters deep in narrow groove, it does not matter if you have 0.3mm step down, but chips will build up in the groove eventualy. You need wider groove to properly evacuate the chips using compressed air or vacuum to prevent jamming and recutting. It is one of the reasons why it's good to use adaptive clearing or at least the trochoidal milling.

> it's easy to get direction from a path

I don't know if the current code of cut operation will properly handle the situation where we have combination.

> make it ergonomic and easy to understand for users.

It is important, that all operations use "conventional milling" by default. that means CCW for outer cuts and CW for inner cuts. And that there is easy way for user to switch to "climb milling" which is the opposite. This caused me a lot of issues when i was new to the CNC game. Climb milling by default on cheap hobby machine caused my parts to be even 1mm or more off. And problem with bCNC was that it always cut CW, which meant that inner cuts were conventional and outer were climb. Which added to the confusion. When i fixed the code to provide consistent conventional milling, my tolerances got to 0.1mm effortlessly. And i don't want any new users to have this issue.

> Allow full recurse, limited to a max (I suggest 100)

i would suggest increasing this limit even more... not sure...

Harvie, 2021-02-13:
I've just found a bug. When i do outside profile of my test file, i get inproperly offset arcs:
[image]

I've checked and this happens even in bCNC version before you added the pocket islands. So can't really blame you, but perhaps you have idea what is going on :-)

It's the same test file as attached to https://github.com/vlachoudis/bCNC/issues/1536#issuecomment-778592306

Harvie, 2021-02-13:
BTW in order to be able to replace original code with your plugin it will be needed to add overcut support.

That would be super cool, because currently bCNC can't do pockets with overcuts. You need to do pocket and the separately profile operation with overcuts, which means that the closest offset is cut twice. Once without overcuts and once with them.  Having pocket and profile merged to single tool would allow us to cut pockets with overcuts in single pass.

MARIOBASZ, 2021-02-13:
With two very close islands, this error occurred

[image]

MARIOBASZ, 2021-02-13:
Or this

[image]

[image]

MARIOBASZ, 2021-02-13:
python would need too much processing time to evaluate island-island overlap maybe?
I'm excited about the advancement of this plugin, Pierre; using as external will surely be an approach to generating paths for pcb (and with two different bit diameters)

DodoLaSaumure, 2021-02-14:
@Harvie 
-  narrow grooves issue.
OK understood

- Cuts with alternate directions
Yes some curious problem of Z jumping in the last pass appear in cuts, we will have to fix them independently from the plugin.
[triangle4.zip](https://github.com/vlachoudis/bCNC/files/5977006/triangle4.zip)

- conventional milling by default
OK understood

- inappropriate arcs in you example
> I've just found a bug. When i do outside profile of my test file, i get improperly offset arcs

Yes I confirm, there is a bug in profile, most probably in offset function but not necessarily (can be in split2contours or remove excluded ). We will have to fix independently from this plugin.

-  it will be needed to add overcut support
Ok understood

@MARIOBASZ 
Thank you for testing. It helps robustness.
In the first early steps of devs of this plugin, I try to address most real life issues. I do not think that the problems you mention appear in most user cases. I do not want to focus on these particular bugs at the moment, because they do not appear to be very easy to reproduce or debug.
But yes later, when we have a minimum stable release of this plugin, we can open issues to try to fix them.
Thanks
Pierre

Harvie, 2021-02-14:
I even have crash sometimes:

```
Traceback (most recent call last):
  File "/home/harvie/Temp/bCNC/bCNC/Utils.py", line 526, in __call__
    return self.func(*args)
  File "/home/harvie/Temp/bCNC/bCNC/ToolsPage.py", line 1747, in execute
    self.tools.getActive().execute(self.app)
  File "/home/harvie/Temp/bCNC/bCNC/plugins/pocketwithislands.py", line 324, in execute
    msg = pocket(blocks,RecursiveDepth,ProfileDir,CutDir,
  File "/home/harvie/Temp/bCNC/bCNC/plugins/pocketwithislands.py", line 73, in pocket
    MyPocket = PocketIsland([path],RecursiveDepth,ProfileDir,CutDir,OverCut, CustomRecursiveDepth,
  File "/home/harvie/Temp/bCNC/bCNC/plugins/pocketwithislands.py", line 137, in __init__
    self.removeInsideIslands()
  File "/home/harvie/Temp/bCNC/bCNC/plugins/pocketwithislands.py", line 236, in removeInsideIslands
    issegin = island.isSegInside(seg)==1
  File "/home/harvie/Temp/bCNC/bCNC/lib/bpath.py", line 1583, in isSegInside
    result = 1 if self.isInside(seg.A) else -1
  File "/home/harvie/Temp/bCNC/bCNC/lib/bpath.py", line 1068, in isInside
    maxx = self.bbox()[2]
  File "/home/harvie/Temp/bCNC/bCNC/lib/bpath.py", line 795, in bbox
    minx = self[0].minx
IndexError: list index out of range

```

DodoLaSaumure, 2021-02-15:
It is hard to debug without the test case (path, parameters of tool, parameters used for doing profile)

here 
```
  File "/home/harvie/Temp/bCNC/bCNC/plugins/pocketwithislands.py", line 236, in removeInsideIslands
    issegin = island.isSegInside(seg)==1
```
```

  File "/home/harvie/Temp/bCNC/bCNC/lib/bpath.py", line 795, in bbox
    minx = self[0].minx
IndexError: list index out of range
```

indicates that island wath an empty path. 
We can add a test before doing 
```python
if len(island)>0 :
isegin = island.isSegInside(seg)==1
```

but I would not add this test before reproducing, to be sure there are no other side effects.
Pierre

DodoLaSaumure, 2021-02-15:
even better would be to test path length in 
```python
	def isSegInside(self,seg):
		if len(self)==0 : 
			return ???
```
I am not sure of what should be the return value here.
I am not sure **return 0** does not produce other bugs outside the func.
Must have test cases + parameters + tool parameters to be sure.

DodoLaSaumure, 2021-02-15:
> I even have crash sometimes

The last exception you sent should be resolved with this commit 
13db370850f3088d812c0ba6376ec737d76ad0cb
( I have tested all the test cases that I had, with no regression seen)

it can be found in PR  #1543

commit b88a0713 (2021-02-13): fix #1536