vlachoudis/bCNC #1540: Bug in profile (confirmed)

DodoLaSaumure, 2021-02-14:
As discussed in #1536 there is a bug in profile function
To reproduce, it appears in making the outside profile of the outside path here
[image]
[profile_bug.ngc.txt](https://github.com/vlachoudis/bCNC/files/5977038/profile_bug.ngc.txt)

Harvie, 2021-02-14:
Fortunately this does not seem to happen when doing profiles using your plugin.

DodoLaSaumure, 2021-02-14:
I have analyzed the code, and the file

cleaned up gcode of the previous example

```
g0 z3
g0 x103.0738 y-15.2312
g0 z0
g2 x100.8334 y-17.5288 i-2.297326 j-0.000979 z0 f300
g1 x2.8371 y-17.5285 z0
g2 x0.6078 y-15.2312 i0.067341 j2.295654 z0
g2 x1.272 y-1.2893 i43.312519 j4.923338 z0
g2 x2.9055 y-0.6078 i1.632867 j-1.615417 z0
g1 x100.8438 y-0.6088 z0
g2 x103.0738 y-2.9055 i-0.066699 j-2.29573 z0
g1 x103.0738 y-15.2312 z0
```

The problem is not is the code neither in the gcode
The problem is when removing ZeroLength segments with precision offset/100.0
The  offset path contains tiny segments which are removed when applying 
removeZeroLength

in bpath.py, line 1289, in offsetClean func

```python
		opath.removeZeroLength(abs(offset)/100.)
```
if you change this line by

```python
		opath.removeZeroLength(abs(offset)/10000.)
```

the problem disappears (tested with a diam 5 bit on the given ngc)

=> Not a bug, but maybe we should consider setting down the zeroLength Segments to less than offset/100.0 I do not know the whole consequences of doing this.

Harvie, 2021-02-14:
in my case abs(offset)/100.  was something like 0.02 mm...
But i still don't understand how this can be a problem...

DodoLaSaumure, 2021-02-15:
This is because the offset path contains tiny elements
opath
block [out,conventional,ccw]:
```
	  0: LINE [105.5737997729978, -15.230134631054636] [105.5738, -15.2312] **L:0.00106537**
	  1: LINE [105.5738, -15.2312] [105.5738, -2.9055] L:12.3257
	  2: LINE [105.5738, -2.9055] [105.57379977726207, -2.9065553149357717] **L:0.00105531**
	  3: CCW  [105.57379977726207, -2.9065553149357717] [100.91640312329129, 1.8901455349183482] C:[100.777, -2.90453] R:4.7967 Phi:[-0.024186..88.3358] L:7.39734
	  4: CCW  [100.91640312329129, 1.8901455349183482] [100.84382552627521, 1.8911999998696816] C:[100.844, -0.6088] R:2.5 Phi:[88.3358..89.9994] L:0.0725878
	  5: LINE [100.84382552627521, 1.8911999998696816] [2.905525526275215, 1.8921999998696815] L:97.9383
	  6: LINE [2.905525526275215, 1.8921999998696815] [2.90618818273605, 1.8921999052809024] L:0.000662656
	  7: CCW  [2.90618818273605, 1.8921999052809024] [-0.5052378713050394, 0.4689450195576075] C:[2.90487, -2.90472] R:4.79692 Phi:[89.9842..135.308] L:3.79458
```
and the reason why it has this tiny elements is because your segments are not perfectly aligned where they should. (arcs are not tangential to the adjacent lines for segments 0 and 2)

DodoLaSaumure, 2021-02-15:
Look at this example
[image]
The green lines are the base profile. They are not perfectly aligned
The red lines are the offset profile.
The blue line is a tiny segment in the offset profile, that appears when linking the offset path.

DodoLaSaumure, 2021-02-15:
These tiny segments appear with length much lower than opath.removeZeroLength(abs(offset)/100.)
I suggest a value of abs(offset)/100 000.

Harvie, 2021-02-15:
I don't really like the idea of increasing this. It probably defies the reason of having it in the first place...
My test file was generated by arcfit feature of bCNC. probably i should check why it is not precise enough....

DodoLaSaumure, 2021-02-15:
Yes i think that it is more appropriate